espressif/esp-iot-solution #434: UVC example gets wrong aspect ratio, always 1:1

suiyi1420, 2024-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I encountered a proportion problem when using the usb_camera_mic_spk example. As shown in picture 1, the rightmost side of each captured frame seems to be in a torn state, and the actual effective aspect ratio is 1:1, and the width is Extrusion, no matter I use 320x240 or 640x480, the actual effective display part is 240x240 or 480x480. What is the problem? I only modified ENABLE_UVC_FRAME_RESOLUTION_ANY=0 
DEMO_UVC_FRAME_WIDTH=320
DEMO_UVC_FRAME_HEIGHT=240
can anyone help me?
[image]
[image]
[image]
[image]

leeebo, 2024-11-27:
@suiyi1420 The USB driver doesn't change the image, including the aspect ratio. It just fetches and then outputs the image as it is.

Please attach your log file to further find out the cause.

suiyi1420, 2024-11-28:
> @suiyi1420 The USB driver doesn't change the image, including the aspect ratio. It just fetches and then outputs the image as it is.
> 
> Please attach your log file to further find out the cause.

When I use VLC Player to connect the camera, the image ratio I get is correct. Is there a way to get the computer to run the camera as USB1.1? In order to verify that it is a device problem? The following is the log of ESP-IDF:
ESP-IDF: V5.1.5
MCU: ESP32-S3

**LOGS：**
I (1482) USB_STREAM: UVC Streaming Config Succeed, Version: 1.4.0
I (1482) USB_STREAM: USB streaming callback register succeed
I (1492) USB_STREAM: Pre-alloc ctrl urb succeed, size = 1024
I (1502) USB_STREAM: USB stream task start
I (1532) USB_STREAM: USB Streaming Start Succeed
I (1532) USB_STREAM: Waiting USB Device Connection
I (1782) USB_STREAM: line 151 HCD_PORT_EVENT_CONNECTION
I (1782) USB_STREAM: Action: ACTION_DEVICE_CONNECT
I (1832) USB_STREAM: Resetting Port
I (1892) USB_STREAM: Setting Port FIFO, 1
I (1892) USB_STREAM: USB Speed: full-speed
I (1892) USB_STREAM: ENUM Stage START, Succeed
I (1902) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
I (1912) USB_STREAM: Default pipe endpoint MPS update to 64
I (1912) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Succeed
I (1922) USB_STREAM: ENUM Stage SET_ADDR, Succeed
I (1942) USB_STREAM: ENUM Stage CHECK_ADDR, Succeed
I (1952) USB_STREAM: ENUM Stage GET_FULL_DEV_DESC, Succeed
*** Device descriptor ***
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x5a3
idProduct 0x9230
bNumConfigurations 1
I (1962) USB_STREAM: ENUM Stage CHECK_FULL_DEV_DESC, Succeed
I (1982) USB_STREAM: ENUM Stage GET_SHORT_CONFIG_DESC, Succeed
I (1992) USB_STREAM: ENUM Stage CHECK_SHORT_CONFIG_DESC, Succeed
I (2002) USB_STREAM: ENUM Stage GET_FULL_CONFIG_DESC, Succeed
*** Configuration descriptor ***
wTotalLength 643
bNumInterfaces 2
bConfigurationValue 1
*** Interface Association Descriptor: Video ***
        *** Interface descriptor ***
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
                *** Endpoint descriptor ***
                bEndpointAddress 0x83   EP 3 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 16
                bInterval 6
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 0
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
        *** Class-specific VS Interface Descriptor ***
        bNumFormats 2
        *** VS Format MJPEG Descriptor ***
        bFormatIndex 0x1
        bNumFrameDescriptors 8
        bDefaultFrameIndex 1
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x1
[... 128 lines cut ...]
I (2412) USB_STREAM: Actual MJPEG format index, format index = 1, contains 8 frames
I (2422) USB_STREAM: Actual Frame: 7, width*height: 320*240, frame index = 3
I (2432) USB_STREAM: ENUM Stage CHECK_FULL_CONFIG_DESC, Succeed
I (2452) USB_STREAM: ENUM Stage SET_CONFIG, Succeed
I (2462) uvc_mic_spk_demo: UVC: get frame list size = 8, current = 2
I (2462) uvc_mic_spk_demo:      frame[0] = 640x480
I (2462) uvc_mic_spk_demo:      frame[1] = 352x288
I (2472) uvc_mic_spk_demo:      frame[2] = 320x240
I (2472) uvc_mic_spk_demo:      frame[3] = 176x144
I (2482) uvc_mic_spk_demo:      frame[4] = 160x120
I (2482) uvc_mic_spk_demo:      frame[5] = 800x600
I (2492) uvc_mic_spk_demo:      frame[6] = 1280x720
I (2492) uvc_mic_spk_demo:      frame[7] = 1280x1024
I (2502) uvc_mic_spk_demo: Device connected
I (2502) USB_STREAM: Probe Format(1), Frame(3) 320*240, interval(666666)
I (2512) USB_STREAM: Probe payload size = 512
bFormatIndex: 1
bFrameIndex: 3
dwFrameInterval: 666666
dwMaxPayloadTransferSize: 0
bInterfaceNumber: 0
I (2572) USB_STREAM: dwMaxPayloadTransferSize set = 512, probed = 0
I (2572) USB_STREAM: Sample processing task start
I (2572) USB_STREAM: Set Device Interface = 1, Alt = 3
I (3292) USB_STREAM: USB Device Connected

leeebo, 2024-12-24:
@suiyi1420 Sorry for the late response.
1. This log looks normal, and there are many options on the ESP32-S3, would you please try more other resolution and frame rates?
2. The usb_stream driver fetches and outputs frames as is, which means the camera did output the strange images
3. `Is there a way to get the computer to run the camera as USB1.1？`  Yes, the fastest way is to add a USB isolator between the PC and the S3 (as these adapters usually only work in full-speed mode)