espressif/esp-iot-solution #439: USB Camera 设备描述符解析错误

NuclearRee, 2024-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32S3 UTG USB

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

usb_stream.c

### Steps to reproduce.

使用ESP32S3 OTG USB开发板进行开发，使用官方示例项目usb_camera_mic_spk进行摄像头推流时。接入摄像头后发现usb设备描述符解析与windows的 usbtreeview.exe解析结果不一致，导致识别不到usb设备，无法进行摄像头推流。不一致数据如下图所示
[image]

lijunru-hub, 2024-12-02:
应该是你这个摄像头不支持在 USB 1.1 Full Speed 下工作，可以用接上 USB 隔离器后再抓一下描述符。

It seems that your camera does not support working in USB 1.1 Full Speed mode. You can try connecting it through a USB isolator and then capture the descriptors again.

NuclearRee, 2024-12-02:
> 应该是你这个摄像头不支持在 USB 1.1 Full Speed 下工作，可以用接上 USB 隔离器后再抓一下描述符。
> 
> It seems that your camera does not support working in USB 1.1 Full Speed mode. You can try connecting it through a USB isolator and then capture the descriptors again.

尝试了一下接了一个usbhub后无法识别到设备了
[image]
下面是用usbtreeview解析的文件
[usb.md](https://github.com/user-attachments/files/17971999/usb.md)

lijunru-hub, 2024-12-02:
usb stream 不支持 USB HUB， USB 隔离器不是 USB HUB，是强制设备运行在 FullSpeed 的设备。
看 log 你这个摄像头在 PC 下是走的 HighSpeed 运行的。你得确定这个设备支不支持 USB FullSpeed。
有些摄像头在 HighSpeed 下可以正常工作，但是可能 FullSpeed 下就没有那么多格式选择，甚至不支持
[image]

NuclearRee, 2024-12-02:
> usb stream 不支持 USB HUB， USB 隔离器不是 USB HUB，是强制设备运行在 FullSpeed 的设备。 看 log 你这个摄像头在 PC 下是走的 HighSpeed 运行的。你得确定这个设备支不支持 USB FullSpeed。 有些摄像头在 HighSpeed 下可以正常工作，但是可能 FullSpeed 下就没有那么多格式选择，甚至不支持 [...]

我又查看了一下设备描述符，有全速模式下的描述符，发现就是我们选取的摄像头不支持全速模式下运行。
可以通过修改驱动层来适配HighSpeed的usb设备？硬件层面会有问题吗

NuclearRee, 2024-12-02:
> usb stream 不支持 USB HUB， USB 隔离器不是 USB HUB，是强制设备运行在 FullSpeed 的设备。 看 log 你这个摄像头在 PC 下是走的 HighSpeed 运行的。你得确定这个设备支不支持 USB FullSpeed。 有些摄像头在 HighSpeed 下可以正常工作，但是可能 FullSpeed 下就没有那么多格式选择，甚至不支持 [...]

我又查看了一下设备的那就是我们选取的摄像头不支持全速模式下运行。

可以通过修改驱动层来适配HighSpeed的usb设备？

lijunru-hub, 2024-12-02:
硬件限定的，除非你换 ESP32-P4

closed 2024-12-02 by NuclearRee