espressif/esp-iot-solution #445: [BUG] MDNS with Button

filzek, 2024-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**After searching to understand and debug we found out that using mDNS with Button if we use a GPIO 2 thread lock inside mDNS** 

New SDK 5.3.2 or older does a thread lock in the task while calling mdns_query_ptr, it just halt and lock it.

It happen if yhe Iot solution Button is being used as of [espressif/button](https://components.espressif.com/components/espressif/button/versions/3.4.0) v3.4.0

it crashes the MDNS when the Button add the GPIO 2, so, if the GPIO 2 is used in button the MDNS jsut halt.

  esp_err_t err2 = mdns_query_ptr(service_name, proto, 5000, 100,  &results2);
  
  so it keeps inside forever.we found out that the lock occrus on
  
  esp_err_t mdns_query_generic(const char *name, const char *service, const char *proto, uint16_t type, mdns_query_transmission_type_t transmission_type, uint32_t timeout, size_t max_results, mdns_result_t **results)

at:
  xSemaphoreTake(search->done_semaphore, portMAX_DELAY);
  
  
 The behavior only happen if the GPIO2 is used to create the button

lijunru-hub, 2024-12-10:
Could you provide a minimal reproducible code example?

leeebo, 2024-12-24:
@filzek It seems that these two functions are not related, whether it is a problem of semaphore use at the application layer?

closed 2025-01-06 by leeebo