espressif/esp-iot-solution #448: USB-DONGLE BTH does not show up on Host

MR-VENOM, 2024-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am testing the USB DONGLE BTH example and the program is uploaded sucessfully but i am not able to list the the dongle with hciconfig. Other ble dongle works fine

This is the log from  ``` sudo dmesg | grep -i usb ```

```
[ 4946.334846] usb 1-6: USB disconnect, device number 13
[ 4950.684320] usb 1-6: new full-speed USB device number 14 using xhci_hcd
[ 4951.513340] usb 1-6: New USB device found, idVendor=303a, idProduct=4001, bcdDevice= 1.00
[ 4951.513351] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4951.513354] usb 1-6: Product: Espressif Device
[ 4951.513358] usb 1-6: Manufacturer: Espressif Systems
[ 4951.513361] usb 1-6: SerialNumber: 123456
[ 4951.534267] usb 1-6: can't set config #1, error -32
[ 4971.742581] usb 1-6: USB disconnect, device number 14
[ 4976.552863] usb 1-6: new full-speed USB device number 15 using xhci_hcd
[ 4976.883479] usb 1-6: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 4976.883491] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4976.883495] usb 1-6: Product: USB JTAG/serial debug unit
[ 4976.883499] usb 1-6: Manufacturer: Espressif
[ 4976.883502] usb 1-6: SerialNumber: 70:04:1D:A4:6C:90
[ 4976.898939] usb 1-6: can't set config #1, error -71
[ 4976.899089] usb 1-6: USB disconnect, device number 15
[ 4977.168543] usb 1-6: new full-speed USB device number 16 using xhci_hcd
[ 4977.498650] usb 1-6: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 4977.498661] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4977.498665] usb 1-6: Product: USB JTAG/serial debug unit
[ 4977.498668] usb 1-6: Manufacturer: Espressif
[ 4977.498671] usb 1-6: SerialNumber: 70:04:1D:A4:6C:90
[ 4977.518310] cdc_acm 1-6:1.0: ttyACM0: USB ACM device
[ 5512.143154] usb 1-6: USB disconnect, device number 16
[ 5514.070173] usb 1-6: new full-speed USB device number 17 using xhci_hcd
[ 5514.401773] usb 1-6: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 5514.401785] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5514.401790] usb 1-6: Product: USB JTAG/serial debug unit
[ 5514.401794] usb 1-6: Manufacturer: Espressif
[ 5514.401797] usb 1-6: SerialNumber: 70:04:1D:A4:6C:90
[ 5514.420933] cdc_acm 1-6:1.0: ttyACM0: USB ACM device
[ 5580.435159] usb 1-6: USB disconnect, device number 17
[ 5826.938020] usb 1-8: new full-speed USB device number 18 using xhci_hcd
[ 5827.282906] usb 1-8: New USB device found, idVendor=303a, idProduct=4001, bcdDevice= 1.00
[ 5827.282917] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5827.282922] usb 1-8: Product: Espressif Device
[ 5827.282926] usb 1-8: Manufacturer: Espressif Systems
[ 5827.282929] usb 1-8: SerialNumber: 123456
[ 5827.305107] usb 1-8: can't set config #1, error -32
[ 5831.586020] usb 1-8: USB disconnect, device number 18
[ 5834.933518] usb 1-8: new full-speed USB device number 19 using xhci_hcd
[ 5835.272099] usb 1-8: New USB device found, idVendor=303a, idProduct=4001, bcdDevice= 1.00
[ 5835.272111] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5835.272115] usb 1-8: Product: Espressif Device
[ 5835.272119] usb 1-8: Manufacturer: Espressif Systems
[ 5835.272122] usb 1-8: SerialNumber: 123456
[ 5835.291927] usb 1-8: can't set config #1, error -32
[ 5850.982614] usb 1-8: USB disconnect, device number 19
[ 5853.229823] usb 1-8: new full-speed USB device number 20 using xhci_hcd
[ 5853.560282] usb 1-8: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 5853.560295] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5853.560299] usb 1-8: Product: USB JTAG/serial debug unit
[ 5853.560303] usb 1-8: Manufacturer: Espressif
[ 5853.560306] usb 1-8: SerialNumber: 24:58:7C:5C:E2:7C
[ 5853.578070] cdc_acm 1-8:1.0: ttyACM0: USB ACM device
[ 5884.299314] usb 1-8: USB disconnect, device number 20
[ 5888.534642] usb 1-8: new full-speed USB device number 21 using xhci_hcd
[ 5889.360745] usb 1-8: New USB device found, idVendor=303a, idProduct=4001, bcdDevice= 1.00
[ 5889.360758] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5889.360763] usb 1-8: Product: Espressif Device
[ 5889.360767] usb 1-8: Manufacturer: Espressif Systems
[ 5889.360770] usb 1-8: SerialNumber: 123456
[ 5889.380584] usb 1-8: can't set config #1, error -32
[ 5973.558520] usb 1-8: USB disconnect, device number 21
[ 5975.770972] usb 1-8: new full-speed USB device number 22 using xhci_hcd
[ 5976.109539] usb 1-8: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 5976.109546] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5976.109548] usb 1-8: Product: USB JTAG/serial debug unit
[ 5976.109550] usb 1-8: Manufacturer: Espressif
[ 5976.109551] usb 1-8: SerialNumber: 24:58:7C:5C:E2:7C
[ 5976.125345] cdc_acm 1-8:1.0: ttyACM0: USB ACM device
[ 6025.466385] usb 1-8: USB disconnect, device number 22
[ 6028.759542] usb 1-8: new full-speed USB device number 23 using xhci_hcd
[ 6029.103086] usb 1-8: New USB device found, idVendor=303a, idProduct=4001, bcdDevice= 1.00
[ 6029.103099] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6029.103104] usb 1-8: Product: Espressif Device
[ 6029.103107] usb 1-8: Manufacturer: Espressif Systems
[ 6029.103111] usb 1-8: SerialNumber: 123456
[ 6029.122964] usb 1-8: can't set config #1, error -32
[ 6515.008944] usb 1-7: new full-speed USB device number 24 using xhci_hcd
[ 6515.339072] usb 1-7: New USB device found, idVendor=0a5c, idProduct=22be, bcdDevice= 1.12
[ 6515.339084] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6515.339089] usb 1-7: Product: BCM20702A0
[ 6515.339092] usb 1-7: Manufacturer: Broadcom Corp
[ 6515.339096] usb 1-7: SerialNumber: 000000000000
[ 6537.600974] usb 1-7: USB disconnect, device number 24
[ 6623.301054] usb 1-8: USB disconnect, device number 23
[ 6625.730038] usb 1-7: new full-speed USB device number 25 using xhci_hcd
[ 6626.557708] usb 1-7: New USB device found, idVendor=303a, idProduct=4001, bcdDevice= 1.00
[ 6626.557720] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6626.557725] usb 1-7: Product: Espressif Device
[ 6626.557729] usb 1-7: Manufacturer: Espressif Systems
[ 6626.557732] usb 1-7: SerialNumber: 123456
[ 6626.579094] usb 1-7: can't set config #1, error -32
[ 6739.527836] usb 1-7: USB disconnect, device number 25
[ 6750.943320] usb 1-7: new full-speed USB device number 26 using xhci_hcd
[ 6751.768987] usb 1-7: New USB device found, idVendor=303a, idProduct=4001, bcdDevice= 1.00
[ 6751.768999] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6751.769004] usb 1-7: Product: Espressif Device
[ 6751.769007] usb 1-7: Manufacturer: Espressif Systems
[ 6751.769011] usb 1-7: SerialNumber: 123456
[ 6751.788817] usb 1-7: can't set config #1, error -32

```

Using esp-idf verion 5.2.3.

esp32s3 eye module devkit.

matezoltanfarkas, 2026-08-16:
I had a similar issue with my ESP32 S3 devkit; the esp was however "seen" via lsusb, however, when I executed `btattach` I had in `journalctl:`
Bluetooth: hci1: Opcode 0x0c03 failed: -110

setting the iso alt count to 1 and using the uart usb port in the end fixed my issue!

XionHsieh1025, 2026-08-25:
Hi @MR-VENOM 

We will close this issue for now. If you run into any further problems, please feel free to reopen it or open a new issue.

closed 2026-08-27 by wujiangang