espressif/esp-iot-solution #452: pwm_audio: is it possible to use for mono configurations? how?

doragasu, 2025-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

It is not clear in the documentation if this component supports mono configurations (with a single speaker). I only want to play mono wav files, and it is not clear if this is supported and if affirmative how should one configure this mode. Should I set bot L and R channels to the same GPIO? Should I set one to an invalid value (like -1)?

In the documentation and sources, there is enum `pwm_audio_channel_t` that has as a possible value `PWM_AUDIO_CH_MONO`, but I cannot see this enum used anywhere in the sources.

YanKE01, 2025-01-06:
Hi, @doragasu . Thank you very much for your interest in `pwm_audio`. If you are not using the left channel, you can set `ledc_channel_left `in `pwm_audio_config_t `to -1, and the same applies for the right channel.

doragasu, 2025-01-07:
Thanks for the clarification. Closing.

closed 2025-01-07 by doragasu