espressif/esp-iot-solution #453: mcp23017 component, fun

edaspb, 2025-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.2

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Steps to reproduce.

Hi everyone,

Looks like mcp23017_set_pullup() function doesn't work for mcp23017.
The code

```
esp_err_t mcp23017_set_pullup(mcp23017_handle_t dev, uint16_t pins)
{
    uint8_t data[] = { MCP23017_PORT_A_BYTE(pins), MCP23017_PORT_B_BYTE(pins) };
    return mcp23017_write(dev, MCP23017_REG_GPIOA, sizeof(data), data); //set REG_GPIOA(); REG_GPIOB();
}
```

Doesn't look good for me. Shouldn't we write to MCP23017_REG_GPPUA, MCP23017_REG_GPPUB register here?

YanKE01, 2025-01-06:
HI, @edaspb . Thank you very much for your feedback. We have noticed that the datasheet indicates the pull-up section needs to control the `GPPU register`. Therefore, in the original code, when set to input, the `mcp23017_set_pullup `function does not work. We will update this issue soon, and we will also release the mcp23017 to the [ESP Component Registry](https://components.espressif.com/), allowing you to use the component in a more convenient way.

closed 2025-01-13 by zhanzhaocheng

YanKE01, 2025-01-15:
Hi, @edaspb . [MCP23017](https://components.espressif.com/components/espressif/mcp23017/versions/0.1.0) has been uploaded to the [ESP Component Registry](https://components.espressif.com/). You can easily integrate it into your project.