espressif/esp-iot-solution #454: esp_tinyuf2 example crashes upon reenabling USB-Serial-JTAG.

wormyrocks, 2025-01-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.2

### Espressif SoC revision.

ESP32-S3 v0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

Upon completion of UF2 bootloader, correctly reset the USB-Serial-JTAG peripheral before resetting MCU.

### What is the actual behavior?

Hits a panic handler while trying to reset the peripheral, somewhere inside `usb_new_phy()`.
https://github.com/espressif/esp-iot-solution/blob/master/components/usb/esp_tinyuf2/esp_tinyuf2.c#L116

### Steps to reproduce.

Code is based on the uf2_ota and uf2_nvs examples: (https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/device/usb_uf2_ota; https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/device/usb_uf2_nvs) on an ESP32-S3 custom PCB.

The USB Serial/JTAG peripheral is normally used for all debugging the main firmware, but wanted to add a user-facing USB firmware update mode.

My modified code both allows a uf2 update, and a config.ini file to update NVS parameters.

UF2 code all runs in userspace - no bootloader hooks installed.

There is also code to manipulate debug LEDs: WS2812Bs using the RMT peripheral. All calls to the RMT peripheral are synchronous and linked to the callback handler (no dedicated LED control task).

The crash is difficult to debug because when it occurs, both the TinyUSB console and USJ peripheral are being reset. I just know that we are entering a panic handler. Not sure if it is a watchdog timeout, stack overflow, or some other kind of issue.

And - i can only tell that the panic handler is triggered because I am using __wrap_esp_panic_handler() to blink a red LED when an error occurs.

By placing a GPIO toggle on either side of the call to usb_ne, i can tell that the error occurs in the call to usb_new_phy().

If the device enters the panic handler and reboots, I can see that the USJ peripheral has been reset correctly. If the call to usb_new_phy() on line 116 of esp_tinyuf2.c is disabled, we avoid the panic handler and esp_reset() properly starts the new firmware. However, after booting the new firmware, the USB device no longer enumerates.

If I enter and leave deep sleep from the firmware, the USJ device enumerates properly again.

This occurs on a Windows host.

wormyrocks, 2025-01-07:
BTW: this happens both with and without `CONFIG_ENABLE_UF2_USB_CONSOLE` enabled.

wormyrocks, 2025-01-21:
Thank you! I have another PR coming in for the tiny_uf2 example that should hopefully be helpful as well.

roma-jam, 2025-01-22:
Related to: https://github.com/hathach/tinyusb/issues/2943