espressif/esp-iot-solution #461: esp_lcd_axs15231b lv_display_set_rotation(lvgl_disp, LV_DISP_ROTATION_90); not work but LV_DISP_ROTATION_0 and LV_DISP_ROTATION_180 is OK

bunjongy, 2025-01-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.2.1

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3 With axs15231b LCD

### Steps to reproduce.

lv_display_set_rotation(lvgl_disp, LV_DISP_ROTATION_0); is Work, Screen is normal show

lv_display_set_rotation(lvgl_disp, LV_DISP_ROTATION_90); Not Work Screen not show

lv_display_set_rotation(lvgl_disp, LV_DISP_ROTATION_180);  is Work Screen is normal show

lv_display_set_rotation(lvgl_disp, LV_DISP_ROTATION_270); Not Work Screen not show

### Build Logs.

```plain

```

Lzw655, 2025-01-21:
Hi @bunjongy,

Are you using the examples from this repository? If not, could you please provide your code project? (excluding the `build` and `managed_components` folders).

bunjongy, 2025-01-21:
[esp32_qr_promptpay-master.zip](https://github.com/user-attachments/files/18486511/esp32_qr_promptpay-master.zip)

this is my project

please help me check

Lzw655, 2025-01-21:
As far as I know, the AXS15231B requires LVGL to use the full-refresh mode for proper display. Additionally, I noticed that the LCD in your project has a TE pin. I recommend using the example from this repository: [lcd_with_te](https://github.com/espressif/esp-iot-solution/tree/master/examples/display/lcd/lcd_with_te).