espressif/esp-iot-solution #463: HCD DWC: EP MPS (192) exceeds supported limit (128) and `CONFIG_USB_HOST_HW_BUFFER_BIAS_*` configs are not taken into account

zry98, 2025-01-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

I expected the example [`usb_camera_mic_spk`](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk) to be working.

### What is the actual behavior?

While running the example [`usb_camera_mic_spk`](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk) with a generic UAC adapter, I'm getting errors like `HCD DWC: EP MPS (192) exceeds supported limit (128)` and `usb_stream/usb_host_helpers.c:355 (_usb_pipe_deinit):invalid args`.

Looks like it's caused by the FIFO MPS limits, but seems they couldn't be changed by the one of the [USB host hardware FIFO size biasing configs](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32s3/api-reference/kconfig.html#config-usb-host-hw-buffer-bias) (I made a [wrong issue](https://github.com/espressif/esp-idf/issues/14941) on this in espressif/esp-idf).

### Steps to reproduce.

<details>
<summary>configs set in the `usb_camera_mic_spk` example</summary>

```cpp
#define ENABLE_UVC_CAMERA_FUNCTION        0        /* enable uvc function */
#define ENABLE_UAC_MIC_SPK_FUNCTION       1        /* enable uac mic+spk function */
#if (ENABLE_UVC_CAMERA_FUNCTION)
#define ENABLE_UVC_FRAME_RESOLUTION_ANY   1        /* Using any resolution found from the camera */
#define ENABLE_UVC_WIFI_XFER              0        /* transfer uvc frame to wifi http */
#endif
#if (ENABLE_UAC_MIC_SPK_FUNCTION)
#define ENABLE_UAC_MIC_SPK_LOOPBACK       0        /* transfer mic data to speaker */
```

</details>

### Debug Logs.

```plain
<details open>
<summary>Logs</summary>

I (645) USB_STREAM: Audio control interface = 0
I (646) USB_STREAM: Speaker feature unit = 9
I (646) USB_STREAM:     Support volume control, ch = 6
I (646) USB_STREAM:     Support mute control, ch = 1
I (646) USB_STREAM: Mic feature unit = 10
I (647) USB_STREAM:     Support volume control, ch = 1
I (647) USB_STREAM:     Support mute control, ch = 1
I (647) USB_STREAM: Speaker Interface found, interface = 1
I (647) USB_STREAM:     Endpoint(ISOC) Addr = 0x2, MPS = 192
I (648) USB_STREAM:     Speaker frequency control Not Support
I (648) USB_STREAM: Mic Interface found interface = 2
I (648) USB_STREAM:     Endpoint(ISOC) Addr = 0x82, MPS = 96
I (649) USB_STREAM:     Mic frequency control Not Support
I (649) USB_STREAM: ENUM Stage CHECK_FULL_CONFIG_DESC, Succeed
I (650) USB_STREAM: ENUM Stage SET_CONFIG, Succeed
I (651) uvc_mic_spk_demo: UAC MIC: get frame list size = 1, current = 0
I (652) uvc_mic_spk_demo:        [0] ch_num = 1, bit_resolution = 16, samples_frequence = 48000, samples_frequence_min = 0, samples_frequence_max = 0
I (652) uvc_mic_spk_demo: UAC MIC: use frame[0] ch_num = 1, bit_resolution = 16, samples_frequence = 48000
I (653) uvc_mic_spk_demo: UAC SPK: get frame list size = 1, current = 0
I (653) uvc_mic_spk_demo:        [0] ch_num = 2, bit_resolution = 16, samples_frequence = 48000, samples_frequence_min = 0, samples_frequence_max = 0
W (654) uvc_mic_spk_demo: UAC SPK: only support 1 channel in this example
I (654) uvc_mic_spk_demo: UAC SPK: use frame[0] ch_num = 2, bit_resolution = 16, samples_frequence = 48000
I (655) uvc_mic_spk_demo: Device connected
E (655) HCD DWC: EP MPS (192) exceeds supported limit (128)
E (655) USB_STREAM: /projects/espressif/esp-iot-solution/components/usb/usb_stream/usb_host_helpers.c:306 (_usb_pipe_init):pipe alloc failed
E (656) USB_STREAM: /projects/espressif/esp-iot-solution/components/usb/usb_stream/usb_stream.c:2739 (_usb_stream_handle_task):pipe init failed
I (657) USB_STREAM: usb stream task wait reset
W (1657) USB_STREAM: usb stream task recover, reason: stream error
I (1657) USB_STREAM: Resetting SPK pipe
I (1657) USB_STREAM: Set Device Interface = 1, Alt = 0
E (1663) USB_STREAM: /projects/espressif/esp-iot-solution/components/usb/usb_stream/usb_host_helpers.c:355 (_usb_pipe_deinit):invalid args
I (1664) USB_STREAM: Resetting MIC pipe
I (1664) USB_STREAM: Set Device Interface = 2, Alt = 0
E (1674) USB_STREAM: /projects/espressif/esp-iot-solution/components/usb/usb_stream/usb_host_helpers.c:355 (_usb_pipe_deinit):invalid args
I (1674) uvc_mic_spk_demo: Device disconnected

</details>

<details>
<summary>UAC adapter descriptor</summary>

*** Device descriptor ***
bcdUSB 1.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 8
idVendor 0x0
idProduct 0x0
bNumConfigurations 1

*** Configuration descriptor ***
wTotalLength 218
bNumInterfaces 4
bConfigurationValue 1
        *** Interface descriptor ***
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 0
        bInterfaceClass 0x1 (Audio)
        bInterfaceSubClass 0x1
        *** Audio control header descriptor ***
        bcdADC 0x100
        wTotalLength 71
        bInCollection 2
                Interface number[0] = 1
                Interface number[1] = 2
        *** Audio control input terminal descriptor ***
        bTerminalID 1
        wTerminalType 0x101
        bAssocTerminal 0
        bNrChannels 2
        wChannelConfig 0x0003
        *** Audio control feature unit descriptor ***
        bUnitID 9
        bSourceID 1
        bControlSize 1
        bmaControls[ch0] 0x1
        bmaControls[ch1] 0x2
        bmaControls[ch2] 0x2
        *** Audio control output terminal descriptor ***
        bTerminalID 3
        wTerminalType 0x301
        bAssocTerminal 0
        *** Audio control input terminal descriptor ***
        bTerminalID 2
        wTerminalType 0x201
        bAssocTerminal 0
        bNrChannels 1
        wChannelConfig 0x0001
        *** Audio control feature unit descriptor ***
        bUnitID 10
        bSourceID 2
        bControlSize 1
        bmaControls[ch0] 0x3
        bmaControls[ch1] 0x0
        *** Audio control output terminal descriptor ***
        bTerminalID 4
        wTerminalType 0x101
        bAssocTerminal 0
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 0
        bInterfaceClass 0x1 (Audio)
        bInterfaceSubClass 0x2
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 1
        bNumEndpoints 1
        bInterfaceClass 0x1 (Audio)
        bInterfaceSubClass 0x2
        *** Audio stream general descriptor ***
        bTerminalLink 1
        bDelay 1
        wFormatTag 1
        *** Audio control header descriptor ***
        bFormatType 1
        bNrChannels 2
        bSubframeSize 2
        bBitResolution 16
        bSamFreqType 1
        tSamFreq[0] 48000
                *** Endpoint descriptor ***
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x9        ISOC
                wMaxPacketSize 192
                bInterval 1
        *** Interface descriptor ***
        bInterfaceNumber 2
        bAlternateSetting 0
        bNumEndpoints 0
        bInterfaceClass 0x1 (Audio)
        bInterfaceSubClass 0x2
        *** Interface descriptor ***
        bInterfaceNumber 2
        bAlternateSetting 1
        bNumEndpoints 1
        bInterfaceClass 0x1 (Audio)
        bInterfaceSubClass 0x2
        *** Audio stream general descriptor ***
        bTerminalLink 4
        bDelay 1
        wFormatTag 1
        *** Audio control header descriptor ***
        bFormatType 1
        bNrChannels 1
        bSubframeSize 2
        bBitResolution 16
        bSamFreqType 1
        tSamFreq[0] 48000
                *** Endpoint descriptor ***
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x9        ISOC
                wMaxPacketSize 96
                bInterval 1
        *** Interface descriptor ***
        bInterfaceNumber 3
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x3 (Unknown)
        bInterfaceSubClass 0x0
                *** Endpoint descriptor ***
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 8
                bInterval 1

</details>
```

leeebo, 2025-01-22:
@zry98 If you connect only with an audio device, try this usb_host_uac example https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_audio_player, check if it still exceeds the limits

lijunru-hub, 2025-01-22:
Please see https://github.com/esp-arduino-libs/ESP32_USB_Stream/issues/4#issuecomment-2481818203

zry98, 2025-03-28:
> Please see [esp-arduino-libs/ESP32_USB_Stream#4 (comment)](https://github.com/esp-arduino-libs/ESP32_USB_Stream/issues/4#issuecomment-2481818203)

Hi @lijunru-hub, as I said, changing the buffer bias option has no effect.

zjzhang-cn, 2025-11-12:
Hi @zry98  .Have you solved this problem?