espressif/esp-iot-solution #468: ESP UVC摄像头帧数据被截断

ffalsk, 2025-02-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### 环境/硬件
ESP-IDF 5.4
ESP32S3N16R8
使用示例usb_camera_mic_spk，**并做了如下修改**
```c++
#define ENABLE_UVC_FRAME_RESOLUTION_ANY   0

#define DEMO_UVC_XFER_BUFFER_SIZE (80 * 1024)

uvc_config_t uvc_config = {
    ...
    .xfer_type = UVC_XFER_BULK,
};
```

sdkconfig做了如下修改
```Kconfig
#
# UVC Stream Config
#
...
CONFIG_NUM_BULK_STREAM_URBS=2
CONFIG_NUM_BULK_BYTES_PER_URB=16384
CONFIG_NUM_ISOC_UVC_URBS=3
CONFIG_NUM_PACKETS_PER_URB=300
# end of UVC Stream Config
```

### 问题
相机采用标准UVC协议，压缩mjpeg格式，bulk传输，在windows上测试正常，esp32上出现单帧画面不完整的情况，观察日志发现驱动并未拼接数据而是直接丢掉数据，而且跟设置的uvc bulk segment size in each urb transfer大小有直接关系，一旦单帧图像超过这个大小，就会被截断。

### 日志

#### 启动日志
```txt

I (27) boot: compile time Feb  7 2025 13:28:15
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (27) boot: efuse block revision: v1.3
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (27) boot.esp32s3: SPI Mode       : DIO
I (27) boot.esp32s3: SPI Flash Size : 16MB
I (28) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (28) boot: ## Label            Usage          Type ST Offset   Length
I (28) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (29) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (29) boot:  2 app0             OTA app          00 10 00010000 007f0000
I (30) boot:  3 app1             OTA app          00 11 00800000 007f0000
I (30) boot:  4 eeprom           Unknown data     01 99 00ff0000 00001000
I (30) boot:  5 spiffs           Unknown data     01 82 00ff1000 0000f000
I (31) boot: End of partition table
I (31) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=32ad0h (207568) map
I (68) esp_image: segment 1: paddr=00042af8 vaddr=3fc9d100 size=05810h ( 22544) load
I (73) esp_image: segment 2: paddr=00048310 vaddr=40374000 size=07d08h ( 32008) load
I (80) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=9e028h (647208) map
I (194) esp_image: segment 4: paddr=000ee050 vaddr=4037bd08 size=11324h ( 70436) load
I (210) esp_image: segment 5: paddr=000ff37c vaddr=600fe100 size=0001ch (    28) load
I (220) boot: Loaded app from partition at offset 0x10000
I (220) boot: Disabling RNG early entropy source...
V (221) mmap: after coalescing, 1 regions are left
[... 553 lines cut ...]
I (14074) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 170, width = 480, height = 320, length = 13387, ptr = 0
I (14147) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 171, width = 480, height = 320, length = 11673, ptr = 0
I (14224) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 172, width = 480, height = 320, length = 12455, ptr = 0
I (14300) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 173, width = 480, height = 320, length = 13289, ptr = 0
I (14373) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 174, width = 480, height = 320, length = 12239, ptr = 0
I (14449) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 175, width = 480, height = 320, length = 12760, ptr = 0
I (14523) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 176, width = 480, height = 320, length = 11999, ptr = 0
I (14598) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 177, width = 480, height = 320, length = 12260, ptr = 0
I (14673) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 178, width = 480, height = 320, length = 11998, ptr = 0
I (14748) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 179, width = 480, height = 320, length = 11753, ptr = 0
I (14823) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 180, width = 480, height = 320, length = 11600, ptr = 0
I (14898) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 181, width = 480, height = 320, length = 11691, ptr = 0
I (14972) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 182, width = 480, height = 320, length = 11451, ptr = 0
I (15048) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 183, width = 480, height = 320, length = 11557, ptr = 0
I (15122) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 184, width = 480, height = 320, length = 11407, ptr = 0
I (15198) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 185, width = 480, height = 320, length = 11407, ptr = 0
I (15272) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 186, width = 480, height = 32, length = 11233, ptr = 0
I (15347) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 187, width = 480, height = 320, length = 10915, ptr = 0

```
**运行日志中间段出现数据截断现象**

被截断部分显示为白色

[image]

leeebo, 2025-02-07:
@ffalsk 可以把 log 开成 verbose，看一下拼包的过程。另外 `uvc bulk segment size in each urb transfer` 如何设置足够大，就不会出现底部白色吗？

ffalsk, 2025-02-08:
@leeebo 
因为部分图像大小不超过16384，为了让每帧都溢出，复现问题，我将`uvc bulk segment size in each urb transfer`设置为8192。
以下是截取的部分日志
```txt
I (14218) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 171, width = 480, height = 320, length = 8190, ptr = 0
V (14219) USB_STREAM: Pipe(UVC): XFER_DONE
V (14219) USB_STREAM: zlp=0, lstp=1, req_len=8192, payload_len=3211, first=0x39, second=0xe9
D (14220) USB_STREAM: bogus packet: len = 3211 39 e9...ff d9

V (14291) USB_STREAM: Pipe(UVC): XFER_DONE
V (14292) USB_STREAM: zlp=0, lstp=0, req_len=8192, payload_len=8192, first=0x02, second=0x82
V (14292) USB_STREAM: header=2 info=0x82, payload_len = 8192, last_pts = 0 , last_scr = 0
V (14292) USB_STREAM: reassembling start ...
V (14292) USB_STREAM: uvc payload = ff d8...1f 4e

V (14293) USB_STREAM: uvc swap buffer length = 8190
V (14293) USB_STREAM: GOT LENGTH 8190
I (14293) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 172, width = 480, height = 320, length = 8190, ptr = 0
V (14294) USB_STREAM: Pipe(UVC): XFER_DONE
V (14295) USB_STREAM: zlp=0, lstp=1, req_len=8192, payload_len=3227, first=0x98, second=0xfd
D (14295) USB_STREAM: bogus packet: len = 3227 98 fd...ff d9

V (14366) USB_STREAM: Pipe(UVC): XFER_DONE
V (14366) USB_STREAM: zlp=0, lstp=0, req_len=8192, payload_len=8192, first=0x02, second=0x83
V (14367) USB_STREAM: header=2 info=0x83, payload_len = 8192, last_pts = 0 , last_scr = 0
V (14367) USB_STREAM: reassembling start ...
V (14367) USB_STREAM: uvc payload = ff d8...ad 6b

V (14367) USB_STREAM: uvc swap buffer length = 8190
V (14368) USB_STREAM: GOT LENGTH 8190
I (14368) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 173, width = 480, height = 320, length = 8190, ptr = 0
V (14369) USB_STREAM: Pipe(UVC): XFER_DONE
V (14369) USB_STREAM: zlp=0, lstp=1, req_len=8192, payload_len=3236, first=0x44, second=0xd8
D (14369) USB_STREAM: bogus packet: len = 3236 44 d8...ff d9

V (14441) USB_STREAM: Pipe(UVC): XFER_ONE
V (14441) USB_STREAM: zlp=0, lstp=0, req_len=8192, payload_len=8192, first=0x02, second=0x82
V (14442) USB_STREAM: header=2 info=0x82, payload_len = 8192, last_pts = 0 , last_scr = 0
V (14442) USB_STREAM:reassembling start ...
V (14442) USB_STREAM: uvc payload = ff d8...95 65

V (14443) USB_STREAM: uvc swap buffer length = 8190
V (14443) USB_STREAM: GOT LENGTH 8190
I (14443) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 174, width = 480, height = 320, length = 8190, ptr = 0
V (14444) USB_STREAM: Pipe(UVC): XFER_DONE
V (14444) USB_STREAM: zlp=0, lstp=1, req_len=8192, payload_len=3215, first=0x38, second=0xc4
D (14444) USB_STREAM: bogus packet: len = 3215 38 c4...ff d9

V (14516) USB_STREAM: Pipe(UVC): XFER_DONE
V (14516) USB_STREAM: zlp=0, lstp=0, req_len=8192, payload_len=8192, first=0x02, second=0x83
V (14517) USB_STREAM: header=2 info=0x83, payload_len = 8192, last_pts = 0 , last_scr = 0
V (14517) USB_STREAM: reassembling start ...
V (14517) USB_STREAM: uvc payload = ff d8...65 3f

V (14518) USB_STREAM: uvc swap buffer length = 8190
V (14518) USB_STREAM: GOT LENGTH 8190
I (14518) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 175, width = 480, height = 320, length = 8190, ptr = 0
V (14519) USB_STREAM: Pipe(UVC): XFER_DONE
V (14519) USB_STREAM: zlp=0, lstp=1, req_len=8192, payload_len=3210, first=0xfa, second=0xd6
D (14519) USB_STREAM: bogus packet: len = 3210 fa d6...ff d9

D (14562) USB_STREAM: USB task running 1200
V (14591) USB_STREAM: Pipe(UVC): XFER_DONE
V (14591) USB_STREAM: zlp=0, lstp=0, req_len=8192, payload_len=8192, first=0x02, second=0x82
V (14591) USB_STREAM: header=2 info=0x82, payload_len = 8192, last_pts = 0 , last_scr = 0
V (14592) USB_STREAM: reassembling start ...
V (14592) USB_STREAM: uvc payload = ff d8...33 49

V (14592) USB_STREAM: uvc swap buffer length = 8190
V (14593) USB_STREAM: GOT LENGTH 8190
I (14593) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 176, width = 480, height = 320, length = 8190, ptr = 0
V (14594) USB_STREAM: Pipe(UVC): XFER_DONE
V (14594) USB_STREAM: zlp=0, lstp=1, req_len=8192, payload_len=3281, first=0x27, second=0xfb
D (14594) USB_STREAM: bogus packet: len = 3281 27 fb...ff d9
```

另外，将`uvc bulk segment size in each urb transfer`设置为40000能处理大部分溢出情况，但是在同时存在其他功能时，分配大buffer可能失败

leeebo, 2025-02-10:
这个摄像头输出的图像数据包头部是 `info=0x82` 或 `info=0x83`，均设置了 EOF bit，所以驱动认为这一帧已经收完了

[image]

你可以把配置项 `"USB Stream"-> "UVC Stream Config" -> "Check EOF bit in payload header"` 从 yes 改为 no，这样驱动不检查这个 bit 就不会有问题了

ffalsk, 2025-02-10:
非常感谢，当我关闭了`Check EOF bit in payload header`后，一切工作正常。

closed 2025-02-10 by ffalsk