espressif/esp-iot-solution #472: espressif/led_indicator 1.0.0 requries espressif/led_strip 2.5.5

diplfranzhoepfinger, 2025-02-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

espressif/led_indicator 1.0.0 requries espressif/led_strip 2.5.5

pls make a new Version of espressif/led_indicator which can work with espressif/led_strip 3.0.0

YanKE01, 2025-02-21:
Hi, @diplfranzhoepfinger . We will implement support for led_strip 3.0.0 in the next major version of led_indicator. Thank you very much for your attention to led_indicator.

diplfranzhoepfinger, 2025-02-21:
> Hi, [@diplfranzhoepfinger](https://github.com/diplfranzhoepfinger) . We will implement support for led_strip 3.0.0 in the next major version of led_indicator. Thank you very much for your attention to led_indicator.

Thanks a lot.

POEG1726, 2025-02-23:
> Hi, [@diplfranzhoepfinger](https://github.com/diplfranzhoepfinger) . We will implement support for led_strip 3.0.0 in the next major version of led_indicator. Thank you very much for your attention to led_indicator.

May I get an approximately date?