espressif/esp-iot-solution #475: LED Indicator组件rgb模式占空比输出似乎有问题

kaidegit, 2025-03-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

1.1.0

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board

### Power Supply used.

Battery

### What is the expected behavior?

全亮度闪烁

### What is the actual behavior?

占空比3%左右闪烁

### Steps to reproduce.

测试代码
```c++
// 改为全亮了
static const blink_step_t triple_green_blink[] = {
    {LED_BLINK_RGB, SET_RGB(255, 255, 255), 0},
    {LED_BLINK_HOLD, LED_STATE_ON, 500},
    {LED_BLINK_HOLD, LED_STATE_OFF, 500},
    {LED_BLINK_HOLD, LED_STATE_ON, 500},
    {LED_BLINK_HOLD, LED_STATE_OFF, 500},
    {LED_BLINK_HOLD, LED_STATE_ON, 500},
    {LED_BLINK_HOLD, LED_STATE_OFF, 500},
    {LED_BLINK_STOP, 0, 0},
};

blink_step_t const *led_mode[] = {
    [BLINK_DOUBLE_RED] = NULL,
    [BLINK_TRIPLE_GREEN] = triple_green_blink,
    [BLINK_WHITE_BREATHE_SLOW] = NULL,
    [BLINK_WHITE_BREATHE_FAST] = NULL,
    [BLINK_BLUE_BREATH] = NULL,
    [BLINK_COLOR_HSV_RING] = NULL,
    [BLINK_COLOR_RGB_RING] = NULL,
    [BLINK_MAX] = NULL,
};

    led_indicator_rgb_config_t rgb_config = {
        .is_active_level_high = false,
        .timer_inited = false,
        .timer_num = LEDC_TIMER_0,
        .red_gpio_num = GPIO_NUM_21,
        .green_gpio_num = GPIO_NUM_20,
        .blue_gpio_num = GPIO_NUM_19,
        .red_channel = LEDC_CHANNEL_0,
        .green_channel = LEDC_CHANNEL_1,
        .blue_channel = LEDC_CHANNEL_2,
    };

    const led_indicator_config_t led_i_config = {
        .mode = LED_RGB_MODE,
        .led_indicator_rgb_config = &rgb_config,
        .blink_lists = led_mode,
        .blink_list_num = BLINK_MAX,
    };

    led_handle = led_indicator_create(&led_i_config);
    assert(led_handle != NULL);

    while (1) {
        led_indicator_start(led_handle, BLINK_TRIPLE_GREEN);
        ESP_LOGI(TAG, "start blink: %d", BLINK_TRIPLE_GREEN);
        vTaskDelay(4000 / portTICK_PERIOD_MS);
        led_indicator_stop(led_handle, BLINK_TRIPLE_GREEN);
        ESP_LOGI(TAG, "stop blink: %d", BLINK_TRIPLE_GREEN);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
```
修改了库内部分代码打log
led_rgb.c
```c++
static esp_err_t led_indicator_rgb_set_duty(led_rgb_t *p_rgb, uint32_t rgb[])
{
    esp_err_t ret;
//    rgb[0] = rgb[0] * p_rgb->max_duty / UINT8_MAX;
//    rgb[1] = rgb[1] * p_rgb->max_duty / UINT8_MAX;
//    rgb[2] = rgb[2] * p_rgb->max_duty / UINT8_MAX;
    if (!p_rgb->is_active_level_high) {
        rgb[0] = p_rgb->max_duty - rgb[0];
        rgb[1] = p_rgb->max_duty - rgb[1];
        rgb[2] = p_rgb->max_duty - rgb[2];
    }
    for (int i = 0; i < 3; i++) {
        ret = ledc_set_duty(LEDC_MODE, p_rgb->rgb_channel[i], rgb[i]);
        LED_RGB_CHECK(ESP_OK == ret, "LEDC set duty error", return ret);
        ret = ledc_update_duty(LEDC_MODE, p_rgb->rgb_channel[i]);
        LED_RGB_CHECK(ESP_OK == ret, "LEDC update duty error", return ret);
        ESP_LOGI(TAG, "LEDC set duty %lu, %lu, %lu, max %lu", rgb[0], rgb[1], rgb[2], p_rgb->max_duty);
    }
    return ESP_OK;
}

esp_err_t led_indicator_rgb_set_on_off(void *rgb_handle, bool on_off)
{
    esp_err_t ret;
    LED_RGB_CHECK(NULL != rgb_handle, "rgb_handle pointer invalid", return ESP_ERR_INVALID_ARG);
    led_rgb_t *p_rgb = (led_rgb_t *)rgb_handle;

    if (on_off) {
        uint32_t rgb[3] = {0};
        led_indicator_hsv2rgb(p_rgb->hsv.value, &rgb[0], &rgb[1], &rgb[2]);
        ESP_LOGI(TAG, "conv rgb %lu %lu %lu", rgb[0], rgb[1], rgb[2]);
        ret = led_indicator_rgb_set_duty(p_rgb, rgb);
        LED_RGB_CHECK(ESP_OK == ret, "LEDC set duty error", return ret);
    } else {
        uint32_t rgb[3] = {0, 0, 0};
        ret = led_indicator_rgb_set_duty(p_rgb, rgb);
        LED_RGB_CHECK(ESP_OK == ret, "LEDC set duty error", return ret);
    }

    return ESP_OK;
}
```

### Debug Logs.

```plain
I (5363) main: start blink: 1
I (5373) led_rgb: LEDC set duty 0, 0, 0, max 8191
I (5373) led_rgb: LEDC set duty 0, 0, 0, max 8191
I (5373) led_rgb: LEDC set duty 0, 0, 0, max 8191
I (5373) led_rgb: conv rgb 255 255 255
I (5373) led_rgb: LEDC set duty 7936, 7936, 7936, max 8191
I (5373) led_rgb: LEDC set duty 7936, 7936, 7936, max 8191
I (5373) led_rgb: LEDC set duty 7936, 7936, 7936, max 8191
I (5873) led_rgb: LEDC set duty 8191, 8191, 8191, max 8191
I (5873) led_rgb: LEDC set duty 8191, 8191, 8191, max 8191
I (5873) led_rgb: LEDC set duty 8191, 8191, 8191, max 8191
I (6373) led_rgb: conv rgb 255 255 255
I (6373) led_rgb: LEDC set duty 7936, 7936, 7936, max 8191
I (6373) led_rgb: LEDC set duty 7936, 7936, 7936, max 8191
I (6373) led_rgb: LEDC set duty 7936, 7936, 7936, max 8191
I (6873) led_rgb: LEDC set duty 8191, 8191, 8191, max 8191
I (6873) led_rgb: LEDC set duty 8191, 8191, 8191, max 8191
I (6873) led_rgb: LEDC set duty 8191, 8191, 8191, max 8191
I (7373) led_rgb: conv rgb 255 255 255
I (7373) led_rgb: LEDC set duty 7936, 7936, 7936, max 8191
I (7373) led_rgb: LEDC set duty 7936, 7936, 7936, max 8191
I (7373) led_rgb: LEDC set duty 7936, 7936, 7936, max 8191
I (7873) led_rgb: LEDC set duty 8191, 8191, 8191, max 8191
I (7873) led_rgb: LEDC set duty 8191, 8191, 8191, max 8191
I (7873) led_rgb: LEDC set duty 8191, 8191, 8191, max 8191
I (9363) main: stop blink: 1
```

kaidegit, 2025-03-05:
临时改成这样能正常跑
```c++
esp_err_t led_indicator_rgb_set_on_off(void *rgb_handle, bool on_off)
{
    esp_err_t ret;
    LED_RGB_CHECK(NULL != rgb_handle, "rgb_handle pointer invalid", return ESP_ERR_INVALID_ARG);
    led_rgb_t *p_rgb = (led_rgb_t *)rgb_handle;

    if (on_off) {
//        uint32_t rgb[3] = {0};
        ret = led_indicator_rgb_set_hsv(p_rgb, p_rgb->hsv.value);
//        led_indicator_hsv2rgb(p_rgb->hsv.value, &rgb[0], &rgb[1], &rgb[2]);
//        ESP_LOGI(TAG, "conv rgb %lu %lu %lu", rgb[0], rgb[1], rgb[2]);
//        ret = led_indicator_rgb_set_duty(p_rgb, rgb);
        LED_RGB_CHECK(ESP_OK == ret, "LEDC set duty error", return ret);
    } else {
        uint32_t rgb[3] = {0, 0, 0};
//        ret = led_indicator_rgb_set_rgb(p_rgb, 0);
        ret = led_indicator_rgb_set_duty(p_rgb, rgb);
        LED_RGB_CHECK(ESP_OK == ret, "LEDC set duty error", return ret);
    }

    return ESP_OK;
}
```

kaidegit, 2025-03-05:
@lijunru-hub

lijunru-hub, 2025-03-11:
这是一个bug，感谢你的报告，错误的原因是未将 RGB 的逻辑值 （0-255）放大到实际的 pwm duty 范围（0-8191），将会尽快被修复。

This is a bug. Thank you for your report. The issue is caused by not scaling the RGB logical values (0-255) to the actual PWM duty range (0-8191). It will be fixed as soon as possible.

closed 2025-03-18 by zhanzhaocheng