espressif/esp-iot-solution #476: esp_sensorless_bldc_control使用adc采样方式,电机启动后报看门狗超时,复位

wangda888, 2025-03-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2182-ga6c3a9cbbb

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

External 5V

### What is the expected behavior?

我使用示例代码,只是修改了简单的IO口配置, 编译运行后, 电机启动通过日志看到报了watchdog timeout,我希望无感直流无刷电机能通过adc采样正常运行起来

### What is the actual behavior?

通过日志看到报了watchdog timeout,然后系统重启

### Steps to reproduce.

只需要运行官方的示例代码即可, 注意需要用ADC采样方式,alignment_mode = ALIGNMENT_ADC; 另外一种比较器的过零采样方式我已经测试了不会出现我说的问题, 另外和板子也无关,我使用买来的开发板,不外接任何东西,也会有同样的错误,经过跟踪,貌似是卡在了adc_oneshot_hal.c 148行(while (!adc_oneshot_ll_get_event(event)) 这里)

### Debug Logs.

```plain
ELF file SHA256: 6bec0494f

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376a7c
  #0  0x40376a7c in esp_restart_noos at /Users/wangda/.platformio/packages/framework-espidf/components/esp_system/port/soc/esp32s3/system_internal.c:160

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x21e8
load:0x403c8700,len:0x4
load:0x403c8704,len:0xeb8
load:0x403cb700,len:0x348c
entry 0x403c8968
I (26) boot: ESP-IDF 5.4.0 2nd stage bootloader
I (26) boot: compile time Mar  6 2025 15:32:43
I (26) boot: Multicore bootloader
D (27) bootloader_flash: non-XMC chip detected by SFDP Read (00), skip.
D (32) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
[... 281 lines cut ...]
    if (bldc_control_set_dir(bldc_control_handle, direction) != ESP_OK) {
        return ESP_FAIL;
    }
    ESP_LOGI(TAG, "wangda test");
    bldc_control_start(bldc_control_handle, 300);

    // if (hal_bldc_button_ctrl_init(GPIO_NUM_0) != ESP_OK) {
    //     return ESP_FAIL;
    // }

    // const esp_timer_create_args_t periodic_timer_args = {
    //     .callback = &hal_bldc_timer_cb,
    //     .name = "periodic"
    // };
    // esp_timer_handle_t periodic_bldc_timer;
    // if (esp_timer_create(&periodic_timer_args, &periodic_bldc_timer) != ESP_OK) {
    //     return ESP_FAIL;
    // }

    // if (esp_timer_start_periodic(periodic_bldc_timer, 1000 * 1000) != ESP_OK) {
    //     return ESP_FAIL;
    // }
    
    return ESP_OK;
}

YanKE01, 2025-03-11:
Hi, @wangda888 . 可以提供一下具体的引脚配置吗？我在`ESP-IDF v5.5-dev-2182-ga6c3a9cbbb`上运行[test_apps](https://github.com/espressif/esp-iot-solution/tree/master/components/motor/esp_sensorless_bldc_control/test_apps/bldc_adc)没有出现超时的问题。

wangda888, 2025-03-11:
Hi,@Yan

> Hi, [@wangda888](https://github.com/wangda888) . 可以提供一下具体的引脚配置吗？我在`ESP-IDF v5.5-dev-2182-ga6c3a9cbbb`上运行[test_apps](https://github.com/espressif/esp-iot-solution/tree/master/components/motor/esp_sensorless_bldc_control/test_apps/bldc_adc)没有出现超时的问题。

以下是我的引脚配置,我直接复制了bldc_fan_config.h配置文件, 给您添麻烦了
#pragma once

#ifdef __cplusplus
extern "C" {
#endif

#define DRV_SLEEP_PIN                10
// Bldc pins configuration
#define UPPER_SWITCH_PIN_U           14
#define UPPER_SWITCH_PIN_V           21
#define UPPER_SWITCH_PIN_W           47
#define LOWER_SWITCH_PIN_U           13
#define LOWER_SWITCH_PIN_V           12
#define LOWER_SWITCH_PIN_W           11
// #define COMPARER_PIN_U               7
// #define COMPARER_PIN_V               6
// #define COMPARER_PIN_W               5

// Button pins configuration
#define SETTING_START_PIN            40
#define SETTING_MODE_PIN             39
#define SETTING_SHACKING_HEAD_PIN    38
#define SETTING_TIMER_PIN            37

// Stepper pins configuration
// #define STEPPER_A_PIN                14
// #define STEPPER_B_PIN                21
// #define STEPPER_C_PIN                47
// #define STEPPER_D_PIN                48

// Timing Parameter Configuration
#define MAXINUM_TIMING_COUNT         5
#define SIGNLE_TIMING_HOUR_DURACTION 2

// Bldc Running Configuration
#define BLDC_MAX_SPEED               1000
#define BLDC_MIN_SPEED               300
#define BLDC_MID_SPEED               (int)(BLDC_MIN_SPEED + (BLDC_MAX_SPEED - BLDC_MIN_SPEED) / 3)
#define BLDC_HIGH_SPEED              (int)(BLDC_MIN_SPEED + (BLDC_MAX_SPEED - BLDC_MIN_SPEED) * 2 / 3)
#define BLDC_MAX_RESTART_COUNT       3

#ifdef __cplusplus
}
#endif

wangda888, 2025-03-11:
> Hi, @wangda888 . 可以提供一下具体的引脚配置吗？我在`ESP-IDF v5.5-dev-2182-ga6c3a9cbbb`上运行[test_apps](https://github.com/espressif/esp-iot-solution/tree/master/components/motor/esp_sensorless_bldc_control/test_apps/bldc_adc)没有出现超时的问题。

另外需要注意，要用ADC采样过零的模式，不要用比较器采样模式，比较器模式没有这个问题

YanKE01, 2025-03-12:
adc模式的话最好吧MCPWM的频率降低一点，单次模式的ADC采样有点慢。修改`bldc_usr_cfg.h`的`MCPWM_CLK_SRC`：
```c
#define MCPWM_CLK_SRC (20 * 1000 * 1000) /*!< Number of count ticks within a period time_us = 1,000,000 / MCPWM_CLK_SRC */
```