espressif/esp-iot-solution #479: SENSOR_HUB_MONITOR example crash on runtime on IDF 5.4.0.

MR-VENOM, 2025-03-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Esp32-S3 Devkit 

### Power Supply used.

USB

### What is the expected behavior?

I expected it to run out of the box but  if was resetting continuously.

### What is the actual behavior?

Restarting at iot_sensor_scan().

### Steps to reproduce.

Just directly run the sensor_hub_monitor example.

### Debug Logs.

```plain
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (26) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (26) boot: compile time Mar 11 2025 17:40:35
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (29) boot: efuse block revision: v1.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0e5b4h ( 58804) map
I (99) esp_image: segment 1: paddr=0001e5dc vaddr=3fc93500 size=01a3ch (  6716) load
I (100) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=24864h (149604) map
I (129) esp_image: segment 3: paddr=0004488c vaddr=3fc94f3c size=011d4h (  4564) load
I (131) esp_image: segment 4: paddr=00045a68 vaddr=40374000 size=0f4f0h ( 62704) load
I (147) esp_image: segment 5: paddr=00054f60 vaddr=600fe100 size=0001ch (    28) load
I (153) boot: Loaded app from partition at offset 0x10000
I (153) boot: Disabling RNG early entropy source...
I (165) cpu_start: Multicore app
I (174) cpu_start: Pro cpu start user code
I (174) cpu_start: cpu freq: 160000000 Hz
I (174) app_init: Application information:
I (174) app_init: Project name:     sensors_monitor
I (178) app_init: App version:      3621aa63-dirty
I (183) app_init: Compile time:     Mar 11 2025 17:40:26
I (188) app_init: ELF file SHA256:  51912d351...
I (192) app_init: ESP-IDF:          v5.4-dirty
I (197) efuse_init: Min chip rev:     v0.0
I (200) efuse_init: Max chip rev:     v0.99 
I (204) efuse_init: Chip rev:         v0.1
I (208) heap_init: Initializing. RAM available for dynamic allocation:
I (215) heap_init: At 3FC96C88 len 00052A88 (330 KiB): RAM
I (220) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (225) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (230) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (236) spi_flash: detected chip: generic
I (239) spi_flash: flash io: dio
W (242) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (254) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (262) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (268) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (275) main_task: Started on CPU0
I (285) main_task: Calling app_main()
I (285) Board_Common: Board Init ...
I (285) Board_Common: register board specified callbacks
E (285) i2c.master: this port has not been initialized, please initialize it first
I (295) i2c_bus: i2c0 bus inited
I (295) i2c_bus: I2C Bus V2 Config Succeed, Version: 1.3.0
I (305) Board_Common: i2c bus create succeed
I (305) Board_Common: Board Info: 

BOARD_NAME: ESP32_S3_USB_OTG_EV
BOARD_VENDOR: Espressif
BOARD_URL: https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-usb-otg/index.html

I (325) Board_Common: Board Init Done !
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0xcc000000  PS      : 0x00060330  A0      : 0x8200ae4b  A1      : 0x3fc994b0  
A2      : 0x00000000  A3      : 0x3c034244  A4      : 0x3c0342b8  A5      : 0x3fc99550  
A6      : 0x00000131  A7      : 0x3c03e4d9  A8      : 0x8200bc8c  A9      : 0x3fc99470  
A10     : 0x3fc994c0  A11     : 0x3c0363c8  A12     : 0x3c0365e0  A13     : 0x00000145  
A14     : 0x3c0363c8  A15     : 0x3c036424  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0xcc000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc  
--- 0x400556d5: strlen in ROM
0x400556e5: strlen in ROM

Backtrace: 0x4bfffffd:0x3fc994b0 0x4200ae48:0x3fc994f0 0x42024093:0x3fc99570 0x4037abf9:0x3fc995a0
--- 0x4200ae48: app_main at /home/nikhi/disk/3rd-party/esp-iot-solution/examples/sensors/sensor_hub_monitor/main/app_main.c:125
0x42024093: main_task at /home/nikhi/esp/v5.4/esp-idf/components/freertos/app_startup.c:208
0x4037abf9: vPortTaskWrapper at /home/nikhi/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

### More Information.

I tries the same code with esp -idf v5.2.3 there it works fine.

Applesaregood123, 2025-03-11:
If you change the main venerable into the first text, you can fix it, but if you run the same code with esp -if v5.2.3 and it crashes, and it takes about 2 hours to reboot.

MR-VENOM, 2025-03-12:
@Applesaregood123 sorry i didn't understand the steps can you give me some more details ?

leeebo, 2025-03-12:
@MR-VENOM Thanks for your report, we have reproduced it in IDF v5.4 and found the bug. 

please change [this line](https://github.com/espressif/esp-iot-solution/blob/master/components/sensors/sensor_hub/linker.lf#L13)         

`sensors_detect -> flash_rodata KEEP() SURROUND(_sensor_hub_detect_fn_array)` 
to 
`sensors_detect -> flash_rodata KEEP() ALIGN(4, pre, post) SURROUND(_sensor_hub_detect_fn_array)`

We will fix it very soon!

MR-VENOM, 2025-03-12:
@leeebo Thanks for the reply i will try that out. I am also facing some of the same issue in 5.2.3 but this is littile diffrent I deed a custom sensor to the sensor-hub ip5306.  and when i call the  iot_sensor_scan(); it is throwing this error  even with the above set changes to the linker.

```
I (401) Board_Common: Board Init Done !
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400556d5  PS      : 0x00060430  A0      : 0x82017bc9  A1      : 0x3fc99f20  
--- 0x400556d5: strlen in ROM

A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3fc99ee0  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3fc9a164  A13     : 0x00000000  
A14     : 0x3c034f3e  A15     : 0x3fc9a134  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
--- 0x400556d5: strlen in ROM
0x400556e5: strlen in ROM

Backtrace: 0x400556d2:0x3fc99f20 0x42017bc6:0x3fc99f30 0x420123e5:0x3fc9a250 0x40381019:0x3fc9a280 0x42009b4d:0x3fc9a2e0 0x42008eb7:0x3fc9a320 0x42020f87:0x3fc9a3a0
--- 0x400556d2: strlen in ROM
0x42017bc6: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1521
0x420123e5: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x40381019: esp_log_writev at /home/nikhi/esp/v5.2.3/esp-idf/components/log/log.c:212
 (inlined by) esp_log_write at /home/nikhi/esp/v5.2.3/esp-idf/components/log/log.c:222
0x42009b4d: iot_sensor_scan at /home/nikhi/Desktop/nikhil/SCIKRAFT/components/sensors/sensor_hub/iot_sensor_hub.c:637 (discriminator 1)
0x42008eb7: app_main at /home/nikhi/Desktop/nikhil/SCIKRAFT/main/main.c:151
0x42020f87: main_task at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/app_startup.c:208

```

the senor works if i comment iot_sensor_scan(); and this part of the code.

```
#if 0
    ESP_LOGI(TAG, "Sensor created, Task name = %s, Type = %s, Sensor Name = %s, Mode = %s, Min Delay = %" PRIu32 " ms",
             SENSOR_DEFAULT_TASK_NAME,
             SENSOR_TYPE_STRING[sensor->type],
             sensor->name,
             SENSOR_MODE_STRING[sensor->mode],
             sensor->min_delay);
#endif
```

Adding bellow the implementation

[ip5306.txt](https://github.com/user-attachments/files/19206134/ip5306.txt)

YanKE01, 2025-03-12:
Hi, @MR-VENOM .  Is it possible that `sensor->type` has exceeded the index range of `SENSOR_TYPE_STRING`?  The sensor types supported by `sensor_hub `are currently fixed. I think we need to add a user-defined type.

MR-VENOM, 2025-03-13:
@YanKE01  Yes. That was the issue . Didn't notice that array now it works fine Thank you.

closed 2025-03-18 by zhanzhaocheng