espressif/esp-iot-solution #488: 关于touch_button_sensor的长按问题

l00jj, 2025-03-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

esp32s3

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3模组

### Power Supply used.

USB

### What is the expected behavior?

我预期设计是可以实现长按的一个触控按钮

### What is the actual behavior?

实际并没有实现长按，大概最多是 3 秒左右自动会转为 “释放”

### Steps to reproduce.

1. 基本按照 lot 的手册简要，没有改动参数
2. GPIO3 引脚接 10cm 的杜邦线充当触控区
3. 烧录并触摸杜邦线

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_log.h"
#include "touch_button_sensor.h"
#include "sdkconfig.h"

uint32_t channel_list[] = { 3, }; // gpio list
float channel_threshold[] = { 0.01, }; // 通道变化阈值

touch_button_config_t config = {
    .channel_num = sizeof(channel_list) / sizeof(channel_list[0]),
    .channel_list = channel_list,            // 使用触摸通道 
    .channel_threshold = channel_threshold,       // 变化阈值
    .channel_gold_value = NULL,                  // 无参考值
    .debounce_times = 3,                        // 3 次读数确认
    .skip_lowlevel_init = false                 // 初始化触摸硬件
};

touch_button_handle_t btn_handle = NULL;

void button_state_callback(touch_button_handle_t handle, uint32_t channel, touch_state_t state, void* arg) {
  switch (state) {
  case TOUCH_STATE_ACTIVE:
    printf("按键通道 %lu 被按下\n", channel);
    break;
  case TOUCH_STATE_INACTIVE:
    printf("按键通道 %lu 被释放\n", channel);
    break;
  }
}

void app_main(void) {
  esp_err_t ret = touch_button_sensor_create(&config, &btn_handle, button_state_callback, NULL);

  while (1) {
    touch_button_sensor_handle_events(btn_handle);
    vTaskDelay(pdMS_TO_TICKS(20));  // 20ms 间隔通常足够
  }
}
```

### Debug Logs.

```plain

```

### More Information.

我的疑问主要是两个：
1. 是这个我的这个杜邦线触控系统提供的电容太粗糙导致最多只能实现3秒长按，还是 `touch_button_sensor ` 当前就是有一个触按时长上限？
2. 如果有这个时长上限，能否提供给用户可自定义的空间？

lijunru-hub, 2025-03-28:
导致这个的原因是 baseline 一直往上走，与当前值误差越来越小，最后触发了抬手。touch 支持长按有时长限制，这个时长的限制有一个参数可以调节。并且一般不会直接触摸 touch 的引脚，会加一层隔离，直接触摸的会会导致值跳变很大。

后面会加入一个上电校准的机制，不会让 baseline 上涨的太离谱。可以等待组件更新

leeebo, 2025-04-01:
@l00jj touch_button_sensor v0.2.1  更改了 baseline 的逻辑，可以更新再试一下

closed 2025-06-18 by zhanzhaocheng