espressif/esp-iot-solution #493: no CONFIG_MODEM_USB_IN_EP_ADDR setting in usb_cdc_4g_module, how to use user defined cat 1 module?

achao1104, 2025-03-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Steps to reproduce.

no CONFIG_MODEM_USB_IN_EP_ADDR setting in usb_cdc_4g_module, how to use user defined cat 1 module? 

### Build Logs.

```plain

```

lijunru-hub, 2025-03-31:
In the latest code, users need to configure the `CONFIG_MODEM_USB_ITF` macro based on the device descriptor.  
This interface (`itf`) typically includes a bulk IN endpoint and a bulk OUT endpoint.

achao1104, 2025-04-03:
Thanks for your reply, and do you have document about how to config CONFIG_MODEM_USB_ITF

skipify, 2025-04-09:
> Thanks for your reply, and do you have document about how to config CONFIG_MODEM_USB_ITF

That's exactly what I need。  +1

lijunru-hub, 2025-04-18:
Find the interface number which has two eps （one bulk in, one bulk out）

[image]

lijunru-hub, 2025-04-18:
Please try with all interfaces number.

aiotfactory, 2025-05-19:
once I find the two ports number, how to config CONFIG_MODEM_USB_ITF in codes?

aiotfactory, 2025-05-19:
My module ML307R DL doesn't support at command: ATD*99***1# , but it does support USB, and what else command I can use instead? Thanks.

lijunru-hub, 2025-05-19:
> once I find the two ports number, how to config CONFIG_MODEM_USB_ITF in codes?

You can run `idf.py menuconfig` to modify the value of `CONFIG_MODEM_USB_ITF` in the configuration menu, but you must first set the target to `user_define`.

lijunru-hub, 2025-05-19:
> My module ML307R DL doesn't support at command: ATD_99_**1# , but it does support USB, and what else command I can use instead? Thanks.

`ATD_99_**1#` is a standard AT dial command. Why isn't it supported by your module? I've tested the ML307, and it works with PPP dialing.  
Please provide detailed logs for further analysis.

aiotfactory, 2025-05-20:
The module is 307R DL,  when it executes the followings, the module resposne with "ERROR"

esp_err_t esp_modem_dce_set_data_mode(esp_modem_dce_t *dce, void *param, void *result)

{

    return esp_modem_dce_generic_command(dce, "ATD*99***1#\r", MODEM_COMMAND_TIMEOUT_MODE_CHANGE,

                                         esp_modem_dce_handle_atd_ppp, NULL);

}

aiotfactory, 2025-05-20:
I also tried "ATD*99***1#\r;" and "ATDT*99***1#\r", and it reports the same error, and when I tried AT+MDIALUP=1,1, it did reponse some ip,gateway etc. but it can not go to next step, because the lwip core can not get ip address and won't raise event "got_ip"

flycatac, 2025-05-23:
因为ML307R不支持任何呼叫控制命令，《AT Commands Reference Guide 4G Series V1.8.1》38页有单独说明。

[image]

aiotfactory, 2025-08-21:
It looks your codes can not parse descriptor of sim7670g, and is there any way to set the address as before instead of parsing it?

I (10645) USBH_CDC: iot usbh cdc version: 2.0.1
I (10675) esp-modem-dte: Enable second AT port
I (10676) esp-modem: --------- Modem PreDefined Info ------------------
I (10677) esp-modem: Model: User Defined
I (10681) esp-modem: Modem itf 3
I (10685) esp-modem: Secondary AT itf 0
I (10690) esp-modem: ----------------------------------------------------
E (10697) modem_board: xyz1
E (10701) modem_board: xyz2
E (10704) modem_board: xyz3
E (11090) ENUM: Configuration descriptor larger than control transfer max length
E (11090) ENUM: [0:0] CHECK_SHORT_CONFIG_DESC FAILED