espressif/esp-iot-solution #495: ESP32-s3 gets reset when them A7672s LTE module intialize .

Beardsome, 2025-04-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

 it shows USB connected then reboot the esp32 s3 again. using USB_CDC_4G_MODULE example

Beardsome, 2025-04-02:
IS anyone works with esp32-s3 and A7672S module. I am not able to create wifi hotspot this example code reboot the esp32 s3 when the A7672S gets ready and it print USB connected and then reset the esp32 S3 . reason for reboot is guru mediation  like below .

I (27) boot: compile time Apr  2 2025 17:45:28
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (42) boot.esp32s3: SPI Mode       : QIO
I (46) boot.esp32s3: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (89) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=2563ch (153148) map
I (122) esp_image: segment 1: paddr=00035664 vaddr=3fc9d400 size=04a98h ( 19096) load
I (126) esp_image: segment 2: paddr=0003a104 vaddr=40374000 size=05f14h ( 24340) load
I (131) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a877ch (690044) map
I (236) esp_image: segment 4: paddr=000e87a4 vaddr=40379f14 size=134b4h ( 79028) load
I (251) esp_image: segment 5: paddr=000fbc60 vaddr=600fe100 size=0001ch (    28) load
I (261) boot: Loaded app from partition at offset 0x10000
I (261) boot: Disabling RNG early entropy source...
I (271) cpu_start: Multicore app
I (280) cpu_start: Pro cpu start user code
I (280) cpu_start: cpu freq: 240000000 Hz
I (280) app_init: Application information:
I (280) app_init: Project name:     usb_cdc_4g_module
I (285) app_init: App version:      1
I (288) app_init: Compile time:     Apr  2 2025 17:46:21
I (294) app_init: ELF file SHA256:  b58db7eba...
--- Warning: Checksum mismatch between flashed and built applications. Checksum of built application is 5a8cac22e57d0347abe5ae2d3c44e64cbfed342eb820582f713d80db8d2b2bd4
I (298) app_init: ESP-IDF:          v5.4-dirty
I (302) efuse_init: Min chip rev:     v0.0
I (306) efuse_init: Max chip rev:     v0.99
I (310) efuse_init: Chip rev:         v0.2
I (314) heap_init: Initializing. RAM available for dynamic allocation:
I (320) heap_init: At 3FCA5708 len 00044008 (272 KiB): RAM
I (325) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (330) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (335) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (341) spi_flash: detected chip: boya
I (344) spi_flash: flash io: qio
W (347) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.   
I (360) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (366) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (373) main_task: Started on CPU0
I (392) main_task: Calling app_main()
I (392) led_indicator: LED Indicator Version: 0.3.0
I (392) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (398) led_indicator: blink_lists is null, use default blink list
W (404) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (420) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0xf, is_active_level_high:1, blink_lists:default
I (432) led_indicator: LED Indicator Version: 0.3.0
I (436) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (445) led_indicator: blink_lists is null, use default blink list
W (451) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (467) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x11, is_active_level_high:1, blink_lists:default
I (479) led_indicator: LED Indicator Version: 0.3.0
I (484) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (492) led_indicator: blink_lists is null, use default blink list
W (498) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (514) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x10, is_active_level_high:1, blink_lists:default
I (540) 4g_main: ====================================
I (541) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (541) 4g_main: ====================================
I (543) modem_board: iot_usbh_modem, version: 1.1.2
[... 189 lines cut ...]
Backtrace: 0x420171c9:0x3fcb2530 0x42017512:0x3fcb2580 0x4201932e:0x3fcb25b0 0x42016ed1:0x3fcb25e0
--- 0x420171c9: cdc_start at C:/Users/lenov/.vscode/extensions/espressif.esp-idf-extension-1.9.1/KTRON/usb_cdc_4g_module/managed_components/espressif_iot_usbh_cdc/iot_usbh_cdc.c:547 (discriminator 1)
 (inlined by) cdc_open at C:/Users/lenov/.vscode/extensions/espressif.esp-idf-extension-1.9.1/KTRON/usb_cdc_4g_module/managed_components/espressif_iot_usbh_cdc/iot_usbh_cdc.c:635 (discriminator 1)
[... cut ...]
--- Warning: Checksum mismatch between flashed and built applications. Checksum of built application is 5a8cac22e57d0347abe5ae2d3c44e64cbfed342eb820582f713d80db8d2b2bd4

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403760a4
--- 0x403760a4: apb_backup_dma_lock at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_system/port/soc/esp32s3/apb_backup_dma.c:16

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x170c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xec4
load:0x403cb700,len:0x3170
entry 0x403c8950

Beardsome, 2025-04-04:
i have solved the issue via selcting chip version correctly . but not able to make esp32 s3 to works as esp hotspot using A7672s

Beardsome, 2025-05-05:
can anyone help me

Beardsome, 2025-05-09:
now the iTF is configured but code is stuck here 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x170c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xec4
load:0x403cb700,len:0x318c
entry 0x403c8950
I (27) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader
I (27) boot: compile time May  9 2025 13:00:24
I (27) boot: Multicore bootloader
I (28) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (42) boot.esp32s3: SPI Mode       : QIO
I (46) boot.esp32s3: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (89) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=25b04h (154372) map
I (123) esp_image: segment 1: paddr=00035b2c vaddr=3fc9d600 size=04b34h ( 19252) load
I (127) esp_image: segment 2: paddr=0003a668 vaddr=40374000 size=059b0h ( 22960) load
I (131) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a9240h (692800) map
I (237) esp_image: segment 4: paddr=000e9268 vaddr=403799b0 size=13c0ch ( 80908) load
I (253) esp_image: segment 5: paddr=000fce7c vaddr=600fe100 size=0001ch (    28) load
I (263) boot: Loaded app from partition at offset 0x10000
I (263) boot: Disabling RNG early entropy source...
I (273) cpu_start: Multicore app
I (281) cpu_start: Pro cpu start user code
I (281) cpu_start: cpu freq: 240000000 Hz
I (282) app_init: Application information:
I (282) app_init: Project name:     usb_cdc_4g_module
I (286) app_init: App version:      1
I (290) app_init: Compile time:     May  9 2025 12:53:28
I (295) app_init: ELF file SHA256:  08bb0e50a...
I (299) app_init: ESP-IDF:          v5.4.1-dirty
I (303) efuse_init: Min chip rev:     v0.0
I (307) efuse_init: Max chip rev:     v0.99
I (311) efuse_init: Chip rev:         v0.2
I (315) heap_init: Initializing. RAM available for dynamic allocation:
I (321) heap_init: At 3FCA5A00 len 00043D10 (271 KiB): RAM
I (327) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (332) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (337) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (343) spi_flash: detected chip: boya
I (346) spi_flash: flash io: qio
W (349) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (361) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (367) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (374) main_task: Started on CPU0
I (393) main_task: Calling app_main()
I (393) led_indicator: LED Indicator Version: 0.3.0
I (393) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (399) led_indicator: blink_lists is null, use default blink list
W (405) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (421) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0xf, is_active_level_high:1, blink_lists:default     
I (433) led_indicator: LED Indicator Version: 0.3.0
I (437) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (446) led_indicator: blink_lists is null, use default blink list
W (452) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
[... 240 lines cut ...]
W (76558) modem_board: Modem event! 0
I (81461) esp-netif_lwip-ppp: Connection lost
I (81462) modem_board: IP event! 7
I (81462) modem_board: Modem Disconnect from PPP Server
W (136559) modem_board: Modem Got IP timeout, retry from start!
E (136559) modem_board: Modem state STAGE_WAIT_IP, retry 5, timeout !
W (136559) modem_board: Retry From Start !
I (136563) modem_board: Modem state STAGE_SYNC, Start
I (136762) modem_board: Network Auto reconnecting ...
I (136762) modem_board: Modem state STAGE_SYNC, Success!
W (136762) 4g_main: Modem Board Event: Network disconnected
I (136862) modem_board: Modem state STAGE_CHECK_SIM, Start
I (136962) modem_board: SIM Card Ready
I (136962) modem_board: Modem state STAGE_CHECK_SIM, Success!
I (136962) 4g_main: Modem Board Event: SIM Card Connected
I (137062) modem_board: Modem state STAGE_CHECK_SIGNAL, Start
I (137162) modem_board: Signal quality: rssi=23, ber=99
I (137162) modem_board: Modem state STAGE_CHECK_SIGNAL, Success!
I (137262) modem_board: Modem state STAGE_CHECK_REGIST, Start
I (137362) modem_board: Network registered, Operator: "405857"
I (137362) modem_board: Modem state STAGE_CHECK_REGIST, Success!
I (137462) modem_board: Modem state STAGE_START_PPP, Start
I (137624) modem_board: Modem state STAGE_START_PPP, Success!
I (137724) modem_board: Modem state STAGE_WAIT_IP, Start
W (137724) modem_board: Modem event! 0

lijunru-hub, 2025-05-12:
It’s possible that the inserted SIM card does not meet the requirements of the current 4G module.

lijunru-hub, 2025-05-12:
The ITF configuration of the 4G module looks fine now.

Beardsome, 2025-05-12:
but it is connected to network and shows fair signal strength and also respond to dial up commnd also works fine for call and sms function . any solution or anything how i debug the issue and solve it

lijunru-hub, 2025-05-12:
For specific details, you may need to consult the 4G module manufacturer.

Beardsome, 2025-05-12:
suddenly it shows this erorr while building the code in esp-idf using VS code . tow days ago its building correctly not it shows this error why

 C:/Espressif/usb_cdc_4g_module/managed_components/espressif__iot_usbh_modem/src/esp_modem_usb_dte.c: In function 'esp_modem_dte_new':       
C:/Espressif/usb_cdc_4g_module/managed_components/espressif__iot_usbh_modem/src/esp_modem_usb_dte.c:498:14: error: 'usbh_cdc_event_callbacks_t' has no member named 'revc_data'; did you mean 'recv_data'?
  498 |             .revc_data = _usb_recv_date_cb,
      |              ^~~~~
      |              recv_data
C:/Espressif/usb_cdc_4g_module/managed_components/espressif__iot_usbh_modem/src/esp_modem_usb_dte.c:514:20: error: 'usbh_cdc_event_callbacks_t' has no member named 'revc_data'; did you mean 'recv_data'?
  514 |     dev_config.cbs.revc_data = NULL;
      |                    ^~~~~
      |                    recv_data
ninja: build stopped: subcommand failed.
```

lijunru-hub, 2025-05-12:
A spelling mistake has been fixed. Please pull the latest `esp-iot-solution` code.

Beardsome, 2025-05-12:
and if possible can you please add Quactel EC200U-CN module in the 4G modem list

lijunru-hub, 2025-05-12:
I don't have this 4G module at the moment, but if I get a chance to test it in the future, I'll add support for it.

skipify, 2025-05-12:
> suddenly it shows this erorr while building the code in esp-idf using VS code . tow days ago its building correctly not it shows this error why [...]

I’m encountering the same issue.
ESP-IDF version: 5.3.1
usbh_modem version: 1.1.5

I found that changing revc_data to recv_data at the location where the error occurs allows the code to compile successfully.

    usbh_cdc_device_config_t dev_config = {
        .vid = 0,
        .pid = 0,
        .itf_num = CONFIG_MODEM_USB_ITF,
        .rx_buffer_size = config->rx_buffer_size,
        .tx_buffer_size = config->tx_buffer_size,
        .cbs = {
            .connect = _usb_conn_callback,
            .disconnect = _usb_disconn_callback,
            .revc_data = _usb_recv_date_cb,
            .user_data = esp_dte,
        },
    };