espressif/esp-iot-solution #496: AIR780E itf2 submit

Sentaku1129, 2025-04-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I tested that the ATitf of air780E is 0x7, but when running the following code, the rssi value is incorrect, I hope it can be improved
int rssi = 0, ber = 0;
modem_board_get_signal_quality(&rssi, &ber);
ESP_LOGI(__func__, "rssi=%d, ber=%d", rssi, ber);

Sentaku1129, 2025-04-10:
The following function can be found to obtain the final rssi and ber values of the function, strstr() function will finally find AT return worth starting, but sscanf function will be processed because of newlines cause input truncation, resulting in an exception, so need to make some adjustments, modified to the following fix1

modem_board_get_signal_quality -> esp_modem_dce_get_signal_quality -> esp_modem_dce_common_handle_csq

fx1: sscanf(strstr(line, "+CSQ:"), "%*s%d,%d", &csq->rssi, &csq->ber);

skipify, 2025-04-10:
The ATitf attribute of Air780E, how should it be selected? I tested both 0x03 and 0x07, and both work, but neither can successfully ping.

I (00:00:14.241) esp-modem: --------- Modem PreDefined Info ------------------
I (00:00:14.242) esp-modem: Model: AIR780E
I (00:00:14.246) esp-modem: Modem itf 7
I (00:00:14.250) esp-modem: ----------------------------------------------------
I (00:00:15.051) MODEM_4G_DRIVER: Modem Board Event: USB connected
I (00:00:15.058) modem_board: DTE reconnect, reconnecting ...
Found OUT endpoint: 4
Found IN endpoint: 136
I (00:00:15.068) USBH_CDC: out ep num: 4
I (00:00:15.073) USBH_CDC: Opened cdc device: 1
I (00:00:15.078) USBH_CDC: New device connected, address: 1
I (00:00:16.056) modem_board: reconnect after 5s...
W (00:00:16.226) MODEM_4G_DRIVER: From 114.114.114.114 icmp_seq=1 timeout
I (00:00:17.056) modem_board: reconnect after 4s...
I (00:00:18.056) modem_board: reconnect after 3s...
I (00:00:19.056) modem_board: reconnect after 2s...
I (00:00:20.056) modem_board: reconnect after 1s...
I (00:00:20.056) modem_board: Modem state STAGE_SYNC, Start
I (00:00:20.070) modem_board: Modem state STAGE_SYNC, Success!
I (00:00:24.229) MODEM_4G_DRIVER: rssi=23, ber=0
I (00:00:24.230) MODEM_4G_DRIVER: Ping addr 114.114.114.114 Restart..
E (00:00:24.231) ping_sock: send error=0
W (00:00:26.226) MODEM_4G_DRIVER: From 114.114.114.114 icmp_seq=1 timeout

Sentaku1129, 2025-04-10:
You can try to change an ip try, my computer directly ping 114.114.114.114 can not connect, you try 8.8.8.8 to see?

ammaree, 2025-04-10:
I’m open Monday 09h00 till 12h00 or 15h00 onwards…

Sentaku1129, 2025-04-10:
> > You can try to change an ip try, my computer directly ping 114.114.114.114 can not connect, you try 8.8.8.8 to see?
> 
> It fails the same way as pinging other addresses, and the HTTP request also fails. Also, when I connect my module via USB to the computer, it can directly access the internet through Air780E.
> 
> I (00:00:46.324) MODEM_4G_DRIVER: rssi=24, ber=0 I (00:00:46.325) MODEM_4G_DRIVER: Ping addr 8.8.8.8 Restart.. E (00:00:46.333) ping_sock: send error=0 W (00:00:48.336) MODEM_4G_DRIVER: From 8.8.8.8 icmp_seq=1 timeout
> 
> E (00:01:16.388) esp-tls: couldn't get hostname for :airtest.openluat.com: getaddrinfo() returns 202, addrinfo=0x0 E (00:01:16.389) transport_base: Failed to open a new connection: 32769 E (00:01:16.396) HTTP_CLIENT: Connection failed, sock < 0 E (00:01:16.402) MODEM_4G_DRIVER: GET request failed, error = ESP_ERR_HTTP_CONNECT

You can check the relevant configuration items in the modem; I think your itf data is not correct

Found OUT endpoint: 4

Found IN endpoint: 136

Normally, I think it should be 2 and 134. I don’t have the relevant equipment with me now, so I need to confirm the details for you tomorrow. You can check the itf value in sdkconfig first

There are some problems with the kconfig of Espressif 1.1.4, which needs to be modified by yourself。You can refer to the following

config MODEM_SUPPORT_SECONDARY_AT_PORT
bool "Enable Dual CDC Mode"
depends on (MODEM_TARGET_MC610_EU || MODEM_TARGET_USER || MODEM_TARGET_EG25_GL || MODEM_TARGET_A7670E || MODEM_TARGET_AIR780_E)
default y if MODEM_TARGET_MC610_EU
default y if MODEM_TARGET_EG25_GL
default y if MODEM_TARGET_A7670E
default y if MODEM_TARGET_AIR780_E
default n if MODEM_TARGET_USER

menu "USB CDC interface config"
// visible if MODEM_TARGET_USER

 config MODEM_USB_ITF
 hex "Modem USB CDC interface"
 default 0x00 if MODEM_TARGET_ML302_DNLM
 default 0x04 if MODEM_TARGET_EC600N_CNLC_N06
 default 0x06 if MODEM_TARGET_MC610_EU
 default 0x05 if MODEM_TARGET_A7600C1
 default 0x02 if MODEM_TARGET_EC20_CE
 default 0x03 if MODEM_TARGET_AIR780_E
 default 0x03 if MODEM_TARGET_EG25_GL
 default 0x03 if MODEM_TARGET_YM310_X09
 default 0x02 if MODEM_TARGET_BG96_MA
 default 0x03 if MODEM_TARGET_SIM7600E
 default 0x04 if MODEM_TARGET_A7670E
 default 0x02 if MODEM_TARGET_SIM7070G
 default 0x02 if MODEM_TARGET_SIM7080G
 default 0x00 if MODEM_TARGET_USER
 help
 USB CDC interface (eg.0x00) used for send AT/data to device

 config MODEM_USB_ITF2
 hex "Modem USB CDC interface 2 (AT Only)"
 depends on MODEM_SUPPORT_SECONDARY_AT_PORT
 default 0x01 if MODEM_TARGET_USER
 default 0x05 if MODEM_TARGET_MC610_EU
 default 0x02 if MODEM_TARGET_EG25_GL
 default 0x05 if MODEM_TARGET_A7670E
 default 0x07 if MODEM_TARGET_AIR780_E
 help
 USB CDC interface 2 (eg.0x01) used for sending AT/data to device
 endmenu

you can clean build,then retry open menu

skipify, 2025-04-10:
@Sentaku1129 

Based on the Kconfig configuration instructions, I modified the interface to 0x03 (which I had used previously), but the problem persists. Now, I am completely unsure where the issue lies and don't know if there are other methods to verify it.

I (00:00:19.605) USBH_CDC: iot usbh cdc version: 1.0.3
I (00:00:19.641) esp-modem: --------- Modem PreDefined Info ------------------
I (00:00:19.641) esp-modem: Model: AIR780E
I (00:00:19.643) esp-modem: Modem itf 3
I (00:00:19.647) esp-modem: ----------------------------------------------------
I (00:00:34.143) MODEM_4G_DRIVER: Modem Board Event: USB connected
I (00:00:34.150) modem_board: DTE reconnect, reconnecting ...
Found OUT endpoint: 2
Found IN endpoint: 132
I (00:00:34.162) USBH_CDC: Opened cdc device: 1
I (00:00:34.166) USBH_CDC: New device connected, address: 1
I (00:00:35.155) modem_board: reconnect after 5s...
I (00:00:36.155) modem_board: reconnect after 4s...
I (00:00:37.155) modem_board: reconnect after 3s...
E (00:00:36.468) esp-tls: couldn't get hostname for :airtest.openluat.com: getaddrinfo() returns 202, addrinfo=0x0
E (00:00:36.469) transport_base: Failed to open a new connection: 32769
E (00:00:36.476) HTTP_CLIENT: Connection failed, sock < 0
E (00:00:36.482) MODEM_4G_DRIVER: GET request failed, error = ESP_ERR_HTTP_CONNECT
I (00:00:36.496) MODEM_4G_DRIVER: rssi=-2113877305, ber=1070382816
I (00:00:36.498) MODEM_4G_DRIVER: Ping addr 8.8.8.8 Restart..
E (00:00:36.504) ping_sock: send error=0

Sentaku1129, 2025-04-11:
> [@Sentaku1129](https://github.com/Sentaku1129)
> 
> Based on the Kconfig configuration instructions, I modified the interface to 0x03 (which I had used previously), but the problem persists. Now, I am completely unsure where the issue lies and don't know if there are other methods to verify it.
> 
> I (00:00:19.605) USBH_CDC: iot usbh cdc version: 1.0.3 I (00:00:19.641) esp-modem: --------- Modem PreDefined Info ------------------ I (00:00:19.641) esp-modem: Model: AIR780E I (00:00:19.643) esp-modem: Modem itf 3 I (00:00:19.647) esp-modem: ---------------------------------------------------- I (00:00:34.143) MODEM_4G_DRIVER: Modem Board Event: USB connected I (00:00:34.150) modem_board: DTE reconnect, reconnecting ... Found OUT endpoint: 2 Found IN endpoint: 132 I (00:00:20.318) USBH_CDC: Opened cdc device: 1 I (00:00:20.323) USBH_CDC: New device connected, address: 1 I (00:00:21.305) modem_board: reconnect after 5s... I (00:00:22.305) modem_board: reconnect after 4s... I (00:00:23.305) modem_board: reconnect after 3s... E (00:00:24.267) esp-tls: couldn't get hostname for :airtest.openluat.com: getaddrinfo() returns 202, addrinfo=0x0 E (00:00:24.267) transport_base: Failed to open a new connection: 32769 E (00:00:24.274) HTTP_CLIENT: Connection failed, sock < 0 E (00:00:24.281) MODEM_4G_DRIVER: GET request failed, error = ESP_ERR_HTTP_CONNECT I (00:00:24.292) MODEM_4G_DRIVER: rssi=-2113875789, ber=1070352848 I (00:00:24.296) MODEM_4G_DRIVER: Ping addr 8.8.8.8 Restart.. E (00:00:24.302) ping_sock: send error=0 I (00:00:24.305) modem_board: reconnect after 2s... I (00:00:25.305) modem_board: reconnect after 1s... I (00:00:25.305) modem_board: Modem state STAGE_SYNC, Start I (00:00:25.319) modem_board: Modem state STAGE_SYNC, Success! W (00:00:26.305) MODEM_4G_DRIVER: From 8.8.8.8 icmp_seq=1 timeout E (00:00:34.297) esp-tls: couldn't get hostname for :airtest.openluat.com: getaddrinfo() returns 202, addrinfo=0x0 E (00:00:34.297) transport_base: Failed to open a new connection: 32769 E (00:00:34.304) HTTP_CLIENT: Connection failed, sock < 0 E (00:00:34.311) MODEM_4G_DRIVER: GET request failed, error = ESP_ERR_HTTP_CONNECT I (00:00:34.322) MODEM_4G_DRIVER: rssi=20, ber=0 I (00:00:34.324) MODEM_4G_DRIVER: Ping addr 8.8.8.8 Restart.. E (00:00:34.331) ping_sock: send error=0 W (00:00:36.325) MODEM_4G_DRIVER: From 8.8.8.8 icmp_seq=1 timeout

Can I see the code in your app_main() function?
It may be related to the initialization below. I have encountered it before. It is similar. modem_config.flags needs to be equal to 0.

modem_config_t modem_config = MODEM_DEFAULT_CONFIG();
// modem_config.flags |= MODEM_FLAGS_INIT_NOT_ENTER_PPP;
// modem_config.flags |= MODEM_FLAGS_INIT_NOT_BLOCK;
ESP_LOGI(__func__, "flags =%#X", modem_config.flags);
modem_config.handler = on_modem_event;
modem_board_init(&modem_config);

skipify, 2025-04-11:
@Sentaku1129 

Thank you so much, it's working perfectly now. Commenting out those two lines of code as per your solution did the trick. Thanks again!

I (00:00:15.070) MODEM_4G_DRIVER: Modem Board Event: USB connected
I (00:00:15.077) modem_board: DTE reconnect, reconnecting ...
Found OUT endpoint: 2
Found IN endpoint: 132
I (00:00:15.088) USBH_CDC: Opened cdc device: 1
I (00:00:15.093) USBH_CDC: New device connected, address: 1
I (00:00:16.075) modem_board: reconnect after 5s...
I (00:00:17.075) modem_board: reconnect after 4s...
I (00:00:18.075) modem_board: reconnect after 3s...
I (00:00:19.075) modem_board: reconnect after 2s...
I (00:00:20.075) modem_board: reconnect after 1s...
I (00:00:20.075) modem_board: Modem state STAGE_SYNC, Start
I (00:00:20.089) modem_board: Network Auto reconnecting ...
I (00:00:20.090) modem_board: Modem state STAGE_SYNC, Success!
W (00:00:20.090) MODEM_4G_DRIVER: Modem Board Event: Network disconnected
I (00:00:20.185) modem_board: Modem state STAGE_CHECK_SIM, Start
I (00:00:20.188) modem_board: SIM Card Ready
I (00:00:20.189) modem_board: Modem state STAGE_CHECK_SIM, Success!
I (00:00:20.189) MODEM_4G_DRIVER: Modem Board Event: SIM Card Connected
I (00:00:20.285) modem_board: Modem state STAGE_CHECK_SIGNAL, Start
I (00:00:20.289) modem_board: Signal quality: rssi=16, ber=0
I (00:00:20.289) modem_board: Modem state STAGE_CHECK_SIGNAL, Success!
I (00:00:20.385) modem_board: Modem state STAGE_CHECK_REGIST, Start
I (00:00:20.388) modem_board: Network registered, Operator: "46011"
I (00:00:20.389) modem_board: Modem state STAGE_CHECK_REGIST, Success!
I (00:00:20.495) modem_board: Modem state STAGE_START_PPP, Start
I (00:00:20.558) modem_board: Modem state STAGE_START_PPP, Success!
I (00:00:20.657) modem_board: Modem state STAGE_WAIT_IP, Start
W (00:00:20.657) modem_board: Modem event! 0
I (00:00:20.664) modem_board: IP event! 3
I (00:00:20.664) modem_board: Got IPv6 address fe80:0000:0000:0000:457f:d205:4e32:fc95
I (00:00:20.670) esp-netif_lwip-ppp: Connected
I (00:00:20.676) esp-modem-netif: PPP state changed event 0: (NETIF_PPP_ERRORNONE)
I (00:00:20.686) esp-netif_lwip-ppp: Connected
I (00:00:20.690) modem_board: IP event! 6
I (00:00:20.694) modem_board: Modem Connected to PPP Server
I (00:00:20.700) modem_board: ppp ip: 10.4.10.16, mask: 255.255.255.255, gw: 10.64.64.64
I (00:00:20.709) modem_board: Main DNS: 114.114.114.114
I (00:00:20.715) modem_board: Backup DNS: 223.5.5.5
I (00:00:20.721) modem_board: Modem state STAGE_WAIT_IP, Success!
I (00:00:20.722) esp-modem-netif: PPP state changed event 0: (NETIF_PPP_ERRORNONE)
I (00:00:20.722) MODEM_4G_DRIVER: changed: ap dns addr (auto): 114.114.114.114
I (00:00:20.736) MODEM_4G_DRIVER: Modem Board Event: Network connected
I (00:00:20.743) MODEM_4G_DRIVER: Ping addr 8.8.8.8 Restart..
I (00:00:20.978) MODEM_4G_DRIVER: 64 bytes from 8.8.8.8 icmp_seq=1 ttl=108 time=220 ms

czy-1, 2025-04-20:
The issue arises because the two UARTs (UART1 and UART2) of the Air780E module operate independently. The ​​modem_board_start_up​​ method only disables echo for UART1 while neglecting UART2. When UART1 enters PPP mode, the AT interface switches to UART2, which still retains echo functionality. This results in abnormal data parsing due to unintended echo interference in UART2.

Attempts to resolve this include:
Re-executing ​​modem_board_start_up​​ after entering ​​STAGE_START_PPP​​ status
Explicitly calling ​​dce->set_echo​​ to disable echo during this stage

While these workarounds allow normal parsing of signal acquisition commands, the ideal solution would be modifying ​​modem_board_start_up​​ to disable echo for ​​both UARTs simultaneously​​ in the ​​iot_usbh_modem​​ library (specifically within ​​usbh_modem_board.c​​).
Since this modification involves low-level driver adjustments within the library and cannot be resolved at the user code level, we kindly request the official development team to address this issue in the firmware/driver implementation.

lijunru-hub, 2025-05-09:
@czy-1 Yes, you're right. We will fix this issue as soon as possible.

jianjunchang, 2025-05-24:
hello, i have try the esp32s3 + yed-c100(the new version of yed-c780e), and the log always show"device 1 gone", and then reset.  

I (171571) 4g_main: Modem Board Event: USB connected
I (171571) modem_board: DTE reconnect, reconnecting ...

I (171576) USBH_CDC: Opened cdcI (172582) modem_board: reconnect after 5s...
I (173582) modem_board: reconnect after 4s...
I (174582) modem_board: reconnect after 3s...
I (175582) modem_board: reconnect after 2s...
I (176136) 4g_main: Ping addr 8.8.8.8 Restart..
I (176137) 4g_main: Task dump

I (176137) 4g_main: Load        Stack left      Name    PRI

I (176139) 4g_main: 0.01        1992    ping 2

I (176143) 4g_main: 0.87        216     main 1

I (176148) 4g_main: 99.73       744     IDLE 0

I (176152) 4g_main: 86.18       736     IDLE 0

I (176157) 4g_main: 0.02        2204    tiT 18

I (176162) 4g_main: 0.20        612     modem_daemon 6

I (176167) 4g_main: 0.01        1376    Tmr Svc 1

I (176172) 4g_main: 0.02        940     usb_data_recv 6

I (176177) 4g_main: 0.00        504     ipc1 24

I (176182) 4g_main: 0.08        2076    usb_lib 5

I (176186) 4g_main: 0.23        664     sys_evt 20

I (176191) 4g_main: 12.48       1748    usbh_cdc 5

I (176196) 4g_main: 0.05        3184    esp_timer 22

I (176201) 4g_main: 0.11        4192    wifi 23

I (176206) 4g_main: 0.00        528     ipc0 24

I (176211) 4g_main: 0.00        2924    httpd 5

I (176215) 4g_main: Free heap=183520 Free mini=118784 bigst=177416, internal=175748 bigst=118784
I (176582) modem_board: reconnect after 1s...
I (176582) modem_board: Modem state STAGE_SYNC, Start
W (176592) esp-modem-dte: Not support sending command in ppp mode, please exit ppp mode first
 response timeoutdem_dce: esp_modem_dce_generic_command(26): Command:AT
I (176609) modem_board: Current timeouts: 1 and errors: 0
W (176609) modem_board: Try to exit ppp mode........
E (177386) USBH: Device 1 gone
W (177397) 4g_main: Modem Board Event: USB disconnected
W (178137) 4g_main: From 8.8.8.8 icmp_seq=1 timeout

Found OUT endpoint: 4
Found IN endpoint: 136
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x19d1
idProduct 0x1
bcdDevice 2.00
[... 391 lines cut ...]
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0xa
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x0
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x4    EP 4 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x88   EP 8 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
I (181070) 4g_main: Modem Board Event: USB connected
I (181075) USBH_CDC: Opened cdc device: 1
I (181080) USBH_CDC: New device connected, address: 1
E (181257) USBH: Device 1 gone
W (181268) 4g_main: Modem Board Event: USB disconnected