espressif/esp-iot-solution #497: elf loader doesn't resolve libc symbols

maedoc, 2025-04-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

esp32-c6-devkitc-1 v1.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-c6-devkitc-1 v1.2

### Power Supply used.

USB

### What is the expected behavior?

I modify the build_elf_file_example to use `ESP_LOGW` following the docs, here's my diff in the iot git repo 
```diff
diff --git a/examples/elf_loader/build_elf_file_example/CMakeLists.txt b/examples/elf_loader/build_elf_file_example/CMakeLists.txt
index a82da18e..6f598ce6 100644
--- a/examples/elf_loader/build_elf_file_example/CMakeLists.txt
+++ b/examples/elf_loader/build_elf_file_example/CMakeLists.txt
@@ -8,7 +8,7 @@ include($ENV{IDF_PATH}/tools/cmake/project.cmake)
 project(hello_world)
 
 # Set other components to link to the ELF file
-# e.g: set(ELF_COMPONENTS "log" "esp_wifi")
+set(ELF_COMPONENTS "log") # "esp_wifi")
 
 include(elf_loader)
 project_elf(hello_world)
diff --git a/examples/elf_loader/build_elf_file_example/main/main.c b/examples/elf_loader/build_elf_file_example/main/main.c
index a6620b23..ac75a515 100644
--- a/examples/elf_loader/build_elf_file_example/main/main.c
+++ b/examples/elf_loader/build_elf_file_example/main/main.c
@@ -6,14 +6,18 @@
 
 #include <stdio.h>
 #include <unistd.h>
+#include <esp_log.h>

 static int s_cnt;
 
+static const char TAG[] = "houp";
+
 int try_test(void)
 {
     for (int i = 0; i < 10; i++) {
-        printf(buffer, s_cnt++);
+        // printf(buffer, s_cnt++);
+        ESP_LOGW(TAG, "holy moly %d", s_cnt++);
         sleep(1);
     }
 
diff --git a/examples/elf_loader/elf_loader_example/main/test_riscv.elf b/examples/elf_loader/elf_loader_example/main/test_riscv.elf
index aa2798d2..ef0fb282 100755
Binary files a/examples/elf_loader/elf_loader_example/main/test_riscv.elf and b/examples/elf_loader/elf_loader_example/main/test_riscv.elf differ
```
and the example should still work. 

I didn't understand how to link the libc here. 

### What is the actual behavior?

It builds but then when it runs, the loader doesn't find vprintf
``` 
(259) main_task: Started on CPU0
I (259) main_task: Calling app_main()
I (259) elf_loader: Start to relocate ELF file
I (269) ELF: ELF loader version: 1.0.0
I (269) ELF: Too much padding before segment[2], padding: 4240
I (269) ELF: elf->entry=0x40812196

E (279) ELF: Can't find common vprintf
E (279) elf_loader: Failed to relocate ELF file errno=-88
I (289) main_task: Returned from app_main()
```

### Steps to reproduce.

1. apply diff above to iot elf loader example
2. build elf, cp to loader
3. build loader, flash and monitor

### Debug Logs.

```plain

```

maedoc, 2025-04-14:
ah ha, if I add `vprintf` to `esp_elf_symbol.c` list, i.e.
```diff
     ESP_ELFSYM_EXPORT(printf),
+    ESP_ELFSYM_EXPORT(vprintf),
     ESP_ELFSYM_EXPORT(vfprintf),
```
then the loader seems to be happy for `vprintf`.  then I see I need to add `"freertos" to the `ELF_COMPONENTS`, like so
```cmake
set(ELF_COMPONENTS "log" "freertos")
```
then I get another missing symbol,
```
I (284) ELF: elf->entry=0x40812086
E (284) ELF: Can't find common esp_crosscore_int_send_yield
E (284) elf_loader: Failed to relocate ELF file errno=-88
```

cwespressif, 2025-06-03:
> ah ha, if I add `vprintf` to `esp_elf_symbol.c` list, i.e.
> 
>      ESP_ELFSYM_EXPORT(printf),
> +    ESP_ELFSYM_EXPORT(vprintf),
>      ESP_ELFSYM_EXPORT(vfprintf),
> then the loader seems to be happy for `vprintf`. then I see I need to add `"freertos" to the `ELF_COMPONENTS`, like so
> 
> set(ELF_COMPONENTS "log" "freertos")
> then I get another missing symbol,
> 
> ```
> I (284) ELF: elf->entry=0x40812086
> E (284) ELF: Can't find common esp_crosscore_int_send_yield
> E (284) elf_loader: Failed to relocate ELF file errno=-88
> ```

You need to add the symbols table for functions or variables that are not linked in the ELF application, for specific details, please refer to: https://github.com/espressif/esp-iot-solution/blob/master/examples/elf_loader/elf_loader_example/README.md#generate-symbols-table-based-on-elf-files.
Please note that the `CONFIG_ELF_LOADER_CUSTOMER_SYMBOLS` configuration item is enabled.

closed 2026-10-27 by zhanzhaocheng