espressif/esp-iot-solution #498: Is there support for using ST77916 with i80?

SonOfPear93, 2025-04-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I've recently bought a display that supposedly uses a ST77916 display driver, and the pinout suggests it will only work using the i80 bus.

The readme for esp_lcd_ST77916 claims to only work with SPI & QSPI, does this mean I cannot use my display with esp-idf at the moment?

I have attempted to use the esp_lcd_ST77916 component anyway, and whilst it does not error, nothing is being displayed to the screen. (in the snippet below I am turning off the backlight, elsewhere I am turning it back on after initialising lvgl etc)

Here is a link to the display I have; https://a.aliexpress.com/_EzG0JQG

Here is a snippet of the code I am attempting to use;
```
    ESP_LOGI(TAG, "Initialize I80 bus");
    esp_lcd_i80_bus_handle_t i80_bus = NULL;
    esp_lcd_i80_bus_config_t bus_config = {
        .dc_gpio_num = GPIO_NUM_2,
        .wr_gpio_num = GPIO_NUM_35,
        .clk_src	= LCD_CLK_SRC_DEFAULT,
        .data_gpio_nums =
        {
            GPIO_NUM_34,
            GPIO_NUM_7,
            GPIO_NUM_6,
            GPIO_NUM_38,
            GPIO_NUM_39,
            GPIO_NUM_40,
            GPIO_NUM_41,
            GPIO_NUM_42,
        },
        .bus_width = 8,
        .max_transfer_bytes = 360 * 360 * static_cast<int>(sizeof(uint16_t)),
    };
    ESP_ERROR_CHECK(esp_lcd_new_i80_bus(&bus_config, &i80_bus));
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_i80_config_t io_config = {
        .cs_gpio_num = GPIO_NUM_36,
        .pclk_hz = 6528000,
        .trans_queue_depth = 20,
        .lcd_cmd_bits = 8,
        .lcd_param_bits = 8,
        .dc_levels = {
            .dc_idle_level = 0,
            .dc_cmd_level = 0,
            .dc_dummy_level = 0,
            .dc_data_level = 1,
        },
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_i80(i80_bus, &io_config, &io_handle));

    ESP_LOGI(TAG, "Install ST77916 panel driver");
    esp_lcd_panel_handle_t panel_handle = NULL;
    const esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = GPIO_NUM_1,
        .rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB,     // Implemented by LCD command `36h`
        .bits_per_pixel = 16,    // Implemented by LCD command `3Ah` (16/18)
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_st77916(io_handle, &panel_config, &panel_handle));
    esp_lcd_panel_reset(panel_handle);
    esp_lcd_panel_init(panel_handle);
    esp_lcd_panel_disp_on_off(panel_handle, true);
```

suda-morris, 2025-04-18:
I'm not familiar with ST77916, but [this code](https://github.com/espressif/esp-iot-solution/blob/master/components/display/lcd/esp_lcd_st77916/esp_lcd_st77916.c#L141) implies there're may need some data adjustment to the parameter data as well when using the i80 interface.  @leeebo May help

suda-morris, 2025-04-18:
Also the I80 LCD usually provides one "WR" and one "RD" line. "RD" line shouldn't be left floating.

SonOfPear93, 2025-04-22:
Thanks for the advice, from what I can see in the ST77916 datasheet that code you're referring to you is only necessary when using QSPI because you need to tell the driver how many data lines to watch. 

The relevant QSPI bit is on page 69 (nice). In the I80 section (pages 61-63) i don't think it mentions needing to transmit any extra commands when wiring color data, besides the 0x2C command which is currently being sent anyway.

Also, you are correct about the RD pin, I originally had it connected to a GPIO pin that I wasn't using in code. Since you pointed it out I've found the bit in the datasheet saying to connect it to either VCC or GND. I now have it connected to a 3v3 VCC pin. When I tested using the I80 example in the esp-idf repo, i could see the display flicker random bands of colour when I had the RD pin connected to the same 3v3 pin as the display VCC pin, but when I connected each to their own 3v3 pin, the screen remained blank. Presumably this was just noise from the display VCC pin, and not actual progress?

One thing I'm struggling to understand is how to set the flags on the esp_lcd_panel_io_i80_config_t object. After reading the datasheet a good few times, I think I should set them like this, but I’m not totally sure..

I can’t find anything to suggest whether to set pclk_idle_low to 0 or 1, but I’m fairly sure the others are correct.

```
.cs_active_high = 0,
.reverse_color_bits ‎ = 0,
.swap_color_bytes ‎ = 0,
.pclk_active_neg = 0,
.pclk_idle_low = 0
```

Another thing I’ve tried to do is compare the esp_lcd_77916.c in this repo with the esp_lcd_panel_st7789.c in the esp-idf repo, but besides the long list of init_cmds, there does not seem to be any meaningful differences.

No matter what I’ve tried, I can’t seem to display anything on my screen even when using the example project.

SonOfPear93, 2025-04-24:
here are the data sheets. Is anyone able to see if there is something I am doing wrong, or maybe something in the esp_lcd_st77916.c that is stopping it from working?

[ST77916_SPEC_V1.0_20220830(1)(2)(2)(1).pdf](https://github.com/user-attachments/files/19895667/ST77916_SPEC_V1.0_20220830.1.2.2.1.pdf)

[HD18001C22规格书.pdf](https://github.com/user-attachments/files/19895671/HD18001C22.pdf)

The dev board I'm using is an unexpected maker S3 pro.
Here is how I have the pins from the display connected to the devboard;

| display pin name | display FFC pin number | dev board pin | misc |
|--------|--------|--------|--------|
| backlight anode | 1 | GPIO 5 | |
| backlight cathode | 2 | GND | |
| display ground | 3 | GND | |
| display VCC | 4 | 3v3 | the one marked IN/OUT |
| display reset | 5 | GPIO 1 | |
| data 7 | 6 | GPIO 42 | |
| data 6 | 7 | GPIO 41 | |
| data 5 | 8 | GPIO 40 | |
| data 4 | 9 | GPIO 39 | |
| data 3 | 10 | GPIO 38 | |
| data 2 | 11 | GPIO 6 | |
| data 1 | 12 | GPIO 7 | |
| data 0 | 13 | GPIO 34 | |
| CS | 14 | GPIO 36 | |
| RD | 15 | 3v3 | not the same 3v3 pin as used for display VCC, or TP-VCC. The one marked OUT ONLY |
| WR | 16 | GPIO 35 | |
| RS aka PCLK | 17 | GPIO 2 | |
| TP-SCL | 18 | GPIO 9 | |
| TP-SDA | 19 | GPIO 8 | |
| TP-RST | 20 | GPIO 3 | |
| TP-INT | 21 | GPIO 4 | |
| TP-VCC | 22 | 3v3 | the one marked IN/OUT |

[image]

SonOfPear93, 2025-04-26:
So it turns out this dev board uses 2 separate LDOs for the 3v3 pins, and to enable the LDO2 pin, you have to set GPIO 17 (the EN pin) to high. As I wasn't doing this, and I had the RD pin connected to this LDO 3v3 pin, the RD was still floating. Once I set GPIO 17 to high, and the RD pin was no longer floating, everything started to work!

@suda-morris thanks for the help|

Incase anyone else stumbles on this thread, here's what worked for me.

```
    gpio_config_t bk_gpio_config = {
        .pin_bit_mask = 1ULL << GPIO_NUM_5,
        .mode = GPIO_MODE_OUTPUT
    };
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));

    gpio_config_t en_gpio_config = {
        .pin_bit_mask = 1ULL << GPIO_NUM_17,
        .mode = GPIO_MODE_OUTPUT
    };
    ESP_ERROR_CHECK(gpio_config(&en_gpio_config));
    gpio_set_level(GPIO_NUM_17, 1);

    ESP_LOGI(TAG, "Initialize I80 bus");
    esp_lcd_i80_bus_config_t bus_config = {
        .dc_gpio_num = GPIO_NUM_2,
        .wr_gpio_num = GPIO_NUM_35,
        .clk_src	= LCD_CLK_SRC_DEFAULT,
        .data_gpio_nums =
        {
            GPIO_NUM_34,
            GPIO_NUM_7,
            GPIO_NUM_6,
            GPIO_NUM_38,
            GPIO_NUM_39,
            GPIO_NUM_40,
            GPIO_NUM_41,
            GPIO_NUM_42,
        },
        .bus_width = 8,
        .max_transfer_bytes = 360 * 360 * static_cast<int>(sizeof(uint16_t)),
        .dma_burst_size = 32
    };
    ESP_ERROR_CHECK(esp_lcd_new_i80_bus(&bus_config, &i80_bus));
    esp_lcd_panel_io_i80_config_t io_config = {
        .cs_gpio_num = GPIO_NUM_36,
        .pclk_hz = 40000000,
        .trans_queue_depth = 20,
        .lcd_cmd_bits = 8,
        .lcd_param_bits = 8,
        .dc_levels = {
            .dc_idle_level = 0,
            .dc_cmd_level = 0,
            .dc_dummy_level = 0,
            .dc_data_level = 1,
        },
        .flags = {
            .cs_active_high = 0,
            .reverse_color_bits = 0,
            .swap_color_bytes = 1,
            .pclk_active_neg = 0,
            .pclk_idle_low = 0
        }
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_i80(i80_bus, &io_config, &io_handle));

    ESP_LOGI(TAG, "Install ST77916 panel driver");
    st77916_vendor_config_t vendor_config = {
        .flags = {
            .use_qspi_interface = 0,
        },
    };
    const esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = GPIO_NUM_1,
        .rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB,
        .bits_per_pixel = 16,
        .flags = {
            .reset_active_high = 0
        },
        .vendor_config = &vendor_config,
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_st77916(io_handle, &panel_config, &panel_handle));
    esp_lcd_panel_reset(panel_handle);
    esp_lcd_panel_init(panel_handle);
    esp_lcd_panel_disp_on_off(panel_handle, true);

    // Set up lvgl gubbins
    // setting up the buffers like this
    // will update my comment if I get it working with LV_DISPLAY_RENDER_MODE_DIRECT
    auto buffer_size = (360 * 360 * lv_display_get_color_format(display)) / 48;
    void* buf1 = esp_lcd_i80_alloc_draw_buffer(io_handle, buffer_size, MALLOC_CAP_DMA);
    void* buf2 = esp_lcd_i80_alloc_draw_buffer(io_handle, buffer_size, MALLOC_CAP_DMA);
    assert(buf1);
    assert(buf2);
    lv_display_set_buffers(display, buf1, buf2, buffer_size, LV_DISPLAY_RENDER_MODE_PARTIAL);

    const esp_lcd_panel_io_callbacks_t cbs = {
        .on_color_trans_done = notify_lvgl_flush_ready,
    };

    ESP_ERROR_CHECK(esp_lcd_panel_io_register_event_callbacks(io_handle, &cbs, display));

    // turn on backlight
    gpio_set_level(GPIO_NUM_5, 1);
```

closed 2025-04-26 by SonOfPear93

SonOfPear93, 2025-04-26:
As it turns out, I didn't even need to connect it to a separate 3v3 source. Connecting the display VCC, TP VCC, and RD pins all to the same 3v3 pin on my dev board seems to work fine!

I'm not sure why it was flickering last time I tested it, maybe it was something else wrong with my code at the time..