espressif/esp-iot-solution #505: Unclear how to use CONFIG_USB_DEVICE_UAC_AS_PART

JVKran, 2025-04-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi there, 

I think that I would have to do the following things in order to use the usb_device_uac component in a project that's already using TinyUSB.

1. Set ```CONFIG_USB_DEVICE_UAC_AS_PART=y```.
2. Configure other Kconfig items such as ```CONFIG_UAC_SPEAKER_CHANNEL_NUM```.
3. Define a custom tusb_config.h and include ```"uac_descriptors.h"``` and ```"tusb_config_uac.h"```.
4. Link it with this CMake.

```make

# Determine whether tinyusb is fetched from component registry or from local path
idf_build_get_property(build_components BUILD_COMPONENTS)
if(tinyusb IN_LIST build_components)
    set(tinyusb_name tinyusb) # Local component
else()
set(tinyusb_name espressif__tinyusb) # Managed component
endif()

idf_component_get_property(tusb_lib ${tinyusb_name} COMPONENT_LIB)
target_include_directories(${tusb_lib} PUBLIC "${PROJECT_DIR}/main/tusb" "${UAC_PATH}/tusb_uac")
target_sources(${tusb_lib} PUBLIC "${PROJECT_DIR}/main/tusb/usb_descriptors.c")

cmake_policy(SET CMP0079 NEW)
target_link_libraries(${tusb_lib} PRIVATE ${COMPONENT_LIB})
```

5. Increase CONFIG_TOTAL_LEN with ```CFG_TUD_AUDIO * TUD_AUDIO_DEVICE_DESC_LEN```.
6. Create ITF_NUM's and EP_NUM's.

``` c
enum {
  EPNUM_MSC_OUT = 0x01,
  EPNUM_MSC_IN,
  EPNUM_AUDIO_OUT,
  EPNUM_AUDIO_FB,
  EPNUM_AUDIO_IN,
}

enum {
  ITF_NUM_MSC = 0,
  ITF_NUM_AUDIO_CONTROL,
  ITF_NUM_AUDIO_STREAMING_SPK,
  ITF_NUM_AUDIO_STREAMING_MIC,
  ITF_NUM_TOTAL
};

#define CONFIG_TOTAL_LEN    (TUD_CONFIG_DESC_LEN + TUD_MSC_DESC_LEN + CFG_TUD_AUDIO * TUD_AUDIO_DEVICE_DESC_LEN)

uint8_t const desc_fs_configuration[] = {
    TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, CONFIG_TOTAL_LEN, 0x00, 100),
    TUD_MSC_DESCRIPTOR(ITF_NUM_MSC, 0, EPNUM_MSC_OUT, EPNUM_MSC_IN, 64),
    TUD_AUDIO_DESCRIPTOR(ITF_NUM_AUDIO_CONTROL, 6, EPNUM_AUDIO_OUT, (0x80 | EPNUM_AUDIO_IN), (0x80 | EPNUM_AUDIO_FB)),
};
````
7. Initialize in the main.

``` c
tinyusb_msc_storage_init_sdmmc
tinyusb_msc_storage_mount
tinyusb_driver_install
tusb_cdc_acm_init
uac_device_config_t config = {
        .skip_tinyusb_init = true,
        .output_cb = uac_device_output_cb,
        .cb_ctx = NULL,
        .spk_itf_num = ITF_NUM_AUDIO_STREAMING_SPK,
    };
    uac_device_init(&config);
```

## The problem
However, I keep getting this output. Without the audio endpoint, it works fine!

```
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
usbd_control_xfer_cb 161: ASSERT FAILED
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
usbd_control_xfer_cb 161: ASSERT FAILED
I (00:00:06.767) battery: VBAT = 4124mV, CHRG-STAT = 0.
I (00:00:06.768) main: VBAT = 4124mV, CHRG-STAT = 0.
I (00:00:06.768) battery: VBAT = 4124mV, CHRG-STAT = 0.
I (00:00:06.771) main: VBAT = 4124mV, CHRG-STAT = 0.
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
```

lijunru-hub, 2025-05-08:
First, determine whether you need a microphone. If not, set `UAC_MIC_CHANNEL_NUM` to 0.

Then, in your main project's `tusb_config.h`, make sure to include `#include "tusb_config_uac.h"` to declare the relevant UAC-related macros.

If the microphone is not needed, both `EPNUM_AUDIO_IN` and `ITF_NUM_AUDIO_STREAMING_MIC` can be omitted.

You can modify the descriptor to:

```c
TUD_AUDIO_DESCRIPTOR(ITF_NUM_AUDIO_CONTROL, 6, EPNUM_AUDIO_OUT, 0, (0x80 | EPNUM_AUDIO_FB)),
```

Most likely, the issue is that you missed this step:
> You need to include `#include "tusb_config_uac.h"` in your main project's `tusb_config.h` to declare UAC-related macros.

JVKran, 2025-05-13:
Hi @lijunru-hub, 

I'm still facing the same issue, but after some tweaking, I now get this value.

[image]

Best regards,

Jochem

JVKran, 2025-05-14:
Hi @lijunru-hub,

I've taken the time to create the following repository for you;[ AEGHB-1067-Reproduction](https://github.com/JVKran/AEGHB-1067-Reproduction). There are three commits;

1. The basic MSC example from the ESP-IDF, with the addition of my own ```tusb_config.h``` and ```usb_descriptors```. I've followed [this guide](https://docs.espressif.com/projects/esp-iot-solution/en/latest/usb/usb_overview/tinyusb_development_guide.html). It completely works and can be flashed on any device (as it uses internal SPI flash).
2. This commit only adds in the [usb_device_uac component](https://components.espressif.com/components/espressif/usb_device_uac/versions/1.2.0) so it should support both MSC and UAC. It doesn't however.
3. This commit strips anything that clutters the application in an effort to create a minimal application for UAC. It still doesn't work.

The last two commits both show the exact same logs;
```
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
```

I'd greatly appreciate it if you could take a look and guide me on what to do to make ```CONFIG_USB_DEVICE_UAC_AS_PART``` because as of now, I'd have to report this as a bug and not working.

Best regards,

Jochem

JVKran, 2025-05-14:
```#include "tusb_config_uac.h"``` must be added to tusb_option.h. Then it works. Otherwise CFG_TUD_AUDIO will not propagate fully into the espressif__tinyusb component.

lijunru-hub, 2025-05-15:
[image]
Sorry for the late reply. Yes, you need to enable the UAC-related macro definitions in `tusb_config.h`.

JVKran, 2025-05-15:
Hi @lijunru-hub,

For the third time, I've been aware of that since the beginning as I am able to read documentation and appear to be one of the very few that do. To sum up the times that I told you this;

- In my first message in this issue;

> 3. Define a custom tusb_config.h and include "uac_descriptors.h" and "tusb_config_uac.h".

- In response to your message also stating exactly the above;

> Thanks for your response. However, that is precisely what I meant with Define a custom tusb_config.h and include "uac_descriptors.h" and "tusb_config_uac.h".

In my last response I'm stating that I have to add ```#include "tusb_config_uac.h"``` to **```tusb_option.h```** and not, I repeat **NOT** in ```tusb_config.h``` as I've already been doing so from the start!

When using Espressif's TinyUSB component, the ```CONFIG_TUD_AUDIO``` macro doesn't propagate to the right layers, even when defined in ```tusb_config.h```.

I'm trying to help you with letting you know what fixed my issue, but this is my last attempt of doing so.

Best regards,

Jochem

esa-cos, 2026-09-09:
Hi I would like to know if it is possible to use usb_device_uac with esp_tinyusb, since I have big trouble to get this working. it starts already with the initial build. I also tried the [AEGHB-1067-Reproduction](https://github.com/JVKran/AEGHB-1067-Reproduction) but I was not able to get it building. I'm just curious if this is really a reliable way of implementing uac and msc, or if it should be done manually.