espressif/esp-iot-solution #506: [Question] `button_matrix` vs. `keyboard_button` differences

EFeru, 2025-05-04:
Hi,
Nice work. I am just a bit puzzled, what is the difference between:
https://github.com/espressif/esp-iot-solution/blob/master/components/button/button_matrix.c
AND
https://github.com/espressif/esp-iot-solution/blob/master/components/keyboard_button/keyboard_button.c

Currently I am using the `keyboard_button` and it works. However I see the `button_matrix` has some additional features like double-click, hold, etc

**Question:** Which one should I use for a normal matrix button? Are there any running efficiency differences?

Related: #480

lijunru-hub, 2025-05-08:
Please use `keyboard_button` for keyboard-related key recognition, which requires a specialized circuit and only supports press and release detection.

`button_matrix` supports common matrix keypads but cannot handle multiple simultaneous key presses.

Regarding efficiency, `keyboard_button` is the most performant—it supports low power consumption and offers a detection rate greater than 1000 Hz.

lijunru-hub, 2025-05-08:
The `keyboard_button` library is primarily designed for detecting full key rollover in keyboard matrix setups. If you're building a keyboard or need full key rollover support, you can use this library.

EFeru, 2025-05-09:
Great answer. The `keyboard_button` fits my needs. Thanks for your time!

closed 2025-05-09 by EFeru