espressif/esp-iot-solution #508: A7672S AS PORTABLE WIFI ROUTER USING ESP32-S

Beardsome, 2025-05-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have tried A7672S module and it creates wifi-hotspot but after 5 minutes the module lost internet connection . and never got connected again untill reset it . what could be the resiong anyone knows please help. 

And can I use ec200U instead of A7672S . what chnages in the should I made ?

lijunru-hub, 2025-05-08:
A persistent and unrecoverable network disconnection might be caused by an issue with the 4G modem itself. If you need to switch to a different model, make sure to configure the correct `CONFIG_MODEM_USB_ITF` macro. You can try identifying a usable interface number (ITF) from the USB descriptors—it should include two bulk endpoints.

[image]

lijunru-hub, 2025-05-09:
Select `CONFIG_MODEM_TARGET_USER`, then manually configure `CONFIG_MODEM_USB_ITF`.

[image]

Beardsome, 2025-05-09:
how to findout CONFIG_MODEM_USB_ITF of the ec200u beacuse code is stuck on gpio: tag and device discriptorr not shown