espressif/esp-iot-solution #509: Add esp-xmodem (IEC-286)

djytw, 2025-05-08:
### Is your feature request related to a problem?

I was trying to find a component for xmodem/ymodem supports and found the [`esp-xmodem` repo on gitee](https://gitee.com/esp-components/esp-xmodem). Is it official? Any possibility to add/port it to Github and idf component registry?

### Describe the solution you'd like.

add esp-xmodem to Github and idf component registry

Narukara, 2026-09-10:
Hello, we're in the process of redesigning an X/YMODEM component. See https://github.com/espressif/idf-extra-components/pull/832

Narukara, 2026-09-30:
The new X/YMODEM component is now available, with documentation coming soon.

https://components.espressif.com/components/espressif/xymodem/

leeebo, 2026-10-26:
Thanks for the follow-up and for pointing us to the new component.

The original request was to make an Espressif-maintained X/YMODEM component available on GitHub and through the ESP Component Registry. Espressif has now published [`espressif/xymodem`](https://components.espressif.com/components/espressif/xymodem), maintained in [`espressif/idf-extra-components`](https://github.com/espressif/idf-extra-components). The component was added upstream in [PR #832](https://github.com/espressif/idf-extra-components/pull/832), so the shared official package is now the right place to use and maintain this implementation.

The current v0.1.1 release supports XMODEM checksum, XMODEM-CRC, and XMODEM-1K; YMODEM and YMODEM-G are not implemented yet, as noted in the [component documentation](https://components.espressif.com/components/espressif/xymodem/versions/0.1.1/readme?language=en). Since the original request to publish an Espressif-maintained component is superseded by this official package, I’m closing this issue. If full YMODEM support is still needed, please open a separate request scoped specifically to implementing YMODEM in `espressif/xymodem`.

Thanks!

closed 2026-10-26 by leeebo