espressif/esp-iot-solution #510: 关于 touch_button_sensor GPIO 的调用问题

l00jj, 2025-05-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

esp32s3

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3模组

### Steps to reproduce.

#define CHANNEL_DATA \
  X(3 , 0.01f, BIT0) \
  X(5  , 0.01f, BIT1) \
  X(14  , 0.01f, BIT2)

#define X(gpio, threshold, signal_bit) gpio,
uint32_t channel_list[] = { CHANNEL_DATA }; // gpio list
#undef X

#define X(gpio, threshold, signal_bit) threshold,
float channel_threshold[] = { CHANNEL_DATA }; // 通道变化阈值
#undef X

static const uint64_t _WIFI_CHANGE_SIGNAL_ = BIT0 | BIT1 | BIT2;
static uint64_t signal_mix = 0;

touch_button_config_t config = {
  .channel_num = sizeof(channel_list) / sizeof(channel_list[0]),  // 通道数量
  .channel_list = channel_list,                                   // 使用触摸通道
  .channel_threshold = channel_threshold,                         // 变化阈值
  .channel_gold_value = NULL,                                     // 无参考值
  .debounce_times = 3,                                            // 3 次读数确认
  .skip_lowlevel_init = false                                     // 初始化触摸硬件
};

touch_button_handle_t btn_handle = NULL;

// 使用 X Macro 自动展开每个 case
uint32_t get_signal_bit(int channel) {
    switch (channel) {
        #define X(ch, th, sb) case ch: return sb;
        CHANNEL_DATA
        #undef X
        default: return 0; // 未匹配到任何 channel
    }
}

void button_state_callback(touch_button_handle_t handle, uint32_t channel, touch_state_t state, void* arg) {
  int signal_bit = get_signal_bit(channel);

  if (state == TOUCH_STATE_ACTIVE) {
    led_r(2); //亮灯
    signal_mix |= signal_bit;
  } else { // TOUCH_STATE_INACTIVE
    led_r(0); // 灭灯
    signal_mix ^= signal_bit;
  }
 
}

### Build Logs.

```plain

```

### More Information.

@leeebo @lijunru-hub 

我刚刚试了新版的 v0.2.1，确实是可以长时间触控了，但出现了新问题。
同一份代码，我使用的针脚是 3 , 5 , 14，一开始我代码写的顺序是 14, 3, 5
v 0.2.1 我只能用 3
v 0.1.0 只能用 14

最后我只能回退到 v0.1.0，且编号输入是 3, 5, 14 顺号，3 个脚都正常，不然就是始终有个不行，很奇怪。。。。😂😂😂

除了这个问题，我有个件建议，可以让用户自定义端口对应的返回编码，这样我就不用专门写一个查表函数了，😁，touch_button_config_t config 里面可以加一个映射就好了

lijunru-hub, 2025-05-08:
NOTICE: [1/5] espressif/cmake_utilities (0.5.3)
NOTICE: [2/5] espressif/touch_sensor_fsm (0.7.0)
NOTICE: [3/5] espressif/touch_sensor_lowlevel (0.8.1)
NOTICE: [4/5] touch_button_sensor (0.2.1) (/home/lijunru/esp/esp-iot-solution/components/touch/touch_button_sensor)
NOTICE: [5/5] idf (5.4.1)

在上述版本下，我测试了一下，是可以读取三通道的。你说的只能用 3 是啥意思？

l00jj, 2025-05-08:
@lijunru-hub 
也有可能是我的这个触控系统比较简陋

[image]

从上到下，分别对应 5, 3, 14，我的代码就是上面我贴出来的，在不更改其他任何东西的情况下。
如果我修改下面代码中3个GPIO的顺序
```
#define CHANNEL_DATA \
  X(3 , 0.01f, BIT0) \
  X(5  , 0.01f, BIT1) \
  X(14  , 0.01f, BIT2)
```

v 0.1.0 会出现始终有1 个到2个脚无法使用，也就是触摸没有反应的情况，这个没有反应是GPIO没有打通还是插件问题就不知道了，没有报错，3个脚要同时能用的情况只能是 3,5,14这样顺序输入。。。。

v 0.2.1 解决的之前不能长按的问题，但是3个脚，尤其是 5 ，始终是无反应，什么顺序都是无反应，然后 3 和 14 有时候有一个正常，然后 14 如果不正常会报错，显示baseline出来

lijunru-hub, 2025-05-08:
@leeebo PTAL

leeebo, 2025-05-12:
@l00jj 和组件版本无关，可能因为直接摸杜邦线触发了阈值上限。因为正常情况下，在产品中因为有塑料盖板，触摸变化率不会高于 10%，因此这个组件设置了阈值上限，超过上限会被视为无效的触摸。

你可以改 menuconfig，将这个判断机制关掉  `menuconfig->components->touch button->Maximum positive change ratio (/1000)` 改成 0

closed 2025-06-18 by zhanzhaocheng