espressif/esp-iot-solution #517: usb_rndis_4g_module not work

aiotfactory, 2025-05-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I followed the guide to do the test with esp32s3, and the log is as following, it looks the usb is found, but it can not ping the ip address, I have tested to ping different ip addresses, it is the same issue.

I (835) app_init: Application information:
I (838) app_init: Project name:     usb_rndis_4g_module
I (844) app_init: App version:      40e383f9
I (849) app_init: Compile time:     May 20 2025 10:49:46
I (855) app_init: ELF file SHA256:  78e867368...
I (860) app_init: ESP-IDF:          v5.3.2-dirty
I (866) efuse_init: Min chip rev:     v0.0
I (870) efuse_init: Max chip rev:     v0.99
I (875) efuse_init: Chip rev:         v0.2
I (880) heap_init: Initializing. RAM available for dynamic allocation:
I (887) heap_init: At 3FCA6598 len 00043178 (268 KiB): RAM
I (894) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (900) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (906) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (912) esp_psram: Adding pool of 6901K of PSRAM memory to heap allocator
I (920) spi_flash: detected chip: generic
I (924) spi_flash: flash io: qio
W (928) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (941) sleep: Configure to isolate all GPIO pins in sleep state
I (948) sleep: Enable automatic switching of GPIO sleep configuration
I (955) main_task: Started on CPU0
I (962) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (968) main_task: Calling app_main()
I (972) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (981) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (991) ch423: ch423_init success
I (995) RNDIS_4G_MODULE: gpio ext module is turned on
I (1000) RNDIS_4G_MODULE: 4g module is powered on
I (11006) USBH_CDC: iot usbh cdc version: 1.1.0
I (11036) usbh_rndis: IOT USBH RNDIS Version: 0.1.0
I (11036) iot_eth: IoT ETH Version: 0.1.0
I (11037) usbh_rndis: USB RNDIS network interface init success
I (11042) iot_eth.netif_glue: 00:00:00:00:00:00
I (11047) iot_eth.netif_glue: ethernet attached to netif
I (11418) usbh_rndis: USB RNDIS CDC new device found
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x2ecc
idProduct 0x3012
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 196
bNumInterfaces 5
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xc0
bMaxPower 500mA
*** Interface Association Descriptor ***
[... 212 lines cut ...]
I (12099) usbh_rndis: resp->Status: 0
I (12108) usbh_rndis: rndis set OID_802_3_MULTICAST_LIST success
I (12115) iot_eth: usb_rndis: MAC address: AC:0C:29:A3:9B:6D
I (12121) usbh_rndis: RNDIS connected success
I (12126) RNDIS_4G_MODULE: IOT_ETH_EVENT_CONNECTED
I (12131) iot_eth.netif_glue: eth_action_handle: 0x3c0c3f24, 0x3c0972e4, 2, 0x3fcba3e4, 0x3c0c3e00
I (12101) usbh_rndis: USB RNDIS CDC notification
I (12146) wifi_init: WiFi IRAM OP enabled
I (12151) wifi_init: WiFi RX IRAM OP enabled
I (12157) camera wifi: wifi_init_softap finished.SSID:ESP-RNDIS-4G password:
I (12164) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (12199) wifi:mode : softAP (a0:85:e3:e6:5c:b5)
I (12200) wifi:Total power save buffer number: 8
I (12200) wifi:Init max length of beacon: 752/752
I (12201) wifi:Init max length of beacon: 752/752
I (12206) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (13141) iot_eth.netif_glue: Ethernet Got IP Address
I (13141) iot_eth.netif_glue: ~~~~~~~~~~~
I (13141) iot_eth.netif_glue: ETHIP:192.168.0.100
I (13145) iot_eth.netif_glue: ETHMASK:255.255.255.0
I (13151) iot_eth.netif_glue: ETHGW:192.168.0.1
I (13156) iot_eth.netif_glue: ~~~~~~~~~~~
W (14216) RNDIS_4G_MODULE: From 139.196.106.200 icmp_seq=1 timeout

W (19216) RNDIS_4G_MODULE: From 139.196.106.200 icmp_seq=1 timeout

lijunru-hub, 2025-05-21:
Why are you pinging the address 139.196.106.200?

aiotfactory, 2025-05-21:
We just changed this IP to a fake one to prevent the leakage of personal information. We also tested 8.8.8.8 and other IPs, none of which are reachable. Moreover, although we can connect to WiFi via mobile phone, we still can't access the internet.

lijunru-hub, 2025-05-22:
I bought it from Taobao in China, so I'm not sure if it's available to you. IMEI, ICCID, and CSQ are not relevant in RNDIS applications. If you want to retrieve this information, you should open a separate AT command channel to access it. This can be done through UART or another USB interface.

aiotfactory, 2025-05-23:
I see that the network interface from RNDIS is named ETH. If we also want to use Ethernet simultaneously, and the Ethernet interface is also named ETH, what should we do if there is a conflict between the two?

aiotfactory, 2025-05-26:
I just changed to use a new module of ML307R DL, now the log is as followings, what is wrong with it? thanks.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x19fc
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe8c
load:0x403cb700,len:0x31a8
entry 0x403c8940
I (24) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (24) boot: compile time May 20 2025 16:15:22
I (24) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (31) boot: efuse block revision: v1.3
I (36) qio_mode: Enabling default flash chip QIO
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : QIO
I (50) boot.esp32s3: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=284a0h (165024) map
I (131) esp_image: segment 1: paddr=000384c8 vaddr=3fca0700 size=052d4h ( 21204) load
I (135) esp_image: segment 2: paddr=0003d7a4 vaddr=40378000 size=02874h ( 10356) load
I (139) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8c984h (575876) map
I (231) esp_image: segment 4: paddr=000cc9ac vaddr=4037a874 size=15e0ch ( 89612) load
I (258) boot: Loaded app from partition at offset 0x10000
I (258) boot: Disabling RNG early entropy source...
I (270) octal_psram: ECC is enabled
I (270) octal_psram: vendor id    : 0x0d (AP)ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x19fc
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe8c
load:0x403cb700,len:0x31a8
entry 0x403c8940
I (23) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (24) boot: compile time May 20 2025 16:15:22
I (24) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (31) boot: efuse block revision: v1.3
I (36) qio_mode: Enabling default flash chip QIO
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : QIO
I (50) boot.esp32s3: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=284a0h (165024) map
I (131) esp_image: segment 1: paddr=000384c8 vaddr=3fca0700 size=052d4h ( 21204) load
I (135) esp_image: segment 2: paddr=0003d7a4 vaddr=40378000 size=02874h ( 10356) load
I (139) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8c984h (575876) map
I (231) esp_image: segment 4: paddr=000cc9ac vaddr=4037a874 size=15e0ch ( 89612) load
I (258) boot: Loaded app from partition at offset 0x10000
I (258) boot: Disabling RNG early entropy source...
I (270) octal_psram: ECC is enabled
[... 288 lines cut ...]
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0xa    EP 10 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
E (7197) usbh_rndis: This device is not a RNDIS device
I (7203) USBH_CDC: New device connected, address: 2
E (8209) usbh_rndis: RNDIS auto detect failed
W (8209) RNDIS_4G_MODULE: Failed to start USB RNDIS driver, try again...
E (10210) usbh_rndis: RNDIS auto detect failed
W (10210) RNDIS_4G_MODULE: Failed to start USB RNDIS driver, try again...
E (12211) usbh_rndis: RNDIS auto detect failed
W (12211) RNDIS_4G_MODULE: Failed to start USB RNDIS driver, try again...
E (14212) usbh_rndis: RNDIS auto detect failed
W (14212) RNDIS_4G_MODULE: Failed to start USB RNDIS driver, try again...
E (16213) usbh_rndis: RNDIS auto detect failed
W (16213) RNDIS_4G_MODULE: Failed to start USB RNDIS driver, try again...
E (18214) usbh_rndis: RNDIS auto detect failed
W (18214) RNDIS_4G_MODULE: Failed to start USB RNDIS driver, try again...

aiotfactory, 2025-05-30:
Anyone can help please?

aiotfactory, 2025-05-30:
Do you have any development boards for ESP32-S3 + ML307? If yes, please send me a link, I'd like to buy one and give it a try.

aiotfactory, 2025-05-30:
If not, could you please provide the schematic of your debugging device? We'd like to check the hardware.

lijunru-hub, 2025-06-03:
你可以采购一下我测试的这款，注意，该模组的供电需求较大，可能需要单独 5V 供电
https://item.taobao.com/item.htm?id=799595862916

aiotfactory, 2025-06-03:
多谢，你们自己设计原理图方便提供下么？因为我们是锂电池供电，因此usb_vbus直接用的4g模组的vcc,并没有升压到5v，不知道是否会影响。

lijunru-hub, 2025-06-04:
没有原理图，我们是直接插在了开发板的 USB 口上。使用的 esp32-s3-dev-kit 开发板，对该模组单独进行 5V 供电。

yxmstar, 2025-06-11:
@lijunru-hub 您好，我使用ml307r也出现了无法上网的情况，查看log似乎一切正常，已经完成如下配置了，请问还有哪些地方需要配置
CONFIG_IN_TRANSFER_BUFFER_SIZE=2048
CONFIG_OUT_TRANSFER_BUFFER_SIZE=2048
以下是我的log
---------------------------------------------------------------------------------------------------------
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1918
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe5c
load:0x403cb700,len:0x303c
entry 0x403c893c
I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Jun 11 2025 15:38:55
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=27470h (160880) map
I (129) esp_image: segment 1: paddr=00037498 vaddr=3fc9e400 size=049a8h ( 18856) load
I (133) esp_image: segment 2: paddr=0003be48 vaddr=40378000 size=041d0h ( 16848) load
I (138) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8a168h (565608) map
I (229) esp_image: segment 4: paddr=000ca190 vaddr=4037c1d0 size=121b4h ( 74164) load
I (253) boot: Loaded app from partition at offset 0x10000
I (253) boot: Disabling RNG early entropy source...
I (265) cpu_start: Multicore app
I (274) cpu_start: Pro cpu start user code
I (274) cpu_start: cpu freq: 240000000 Hz
I (274) app_init: Application information:
I (277) app_init: Project name:     usb_rndis_4g_module
I (283) app_init: App version:      140c9545-dirty
I (288) app_init: Compile time:     Jun 11 2025 21:54:31
I (294) app_init: ELF file SHA256:  2f70135b5...
I (299) app_init: ESP-IDF:          v5.3.1-dirty
I (305) efuse_init: Min chip rev:     v0.0
I (310) efuse_init: Max chip rev:     v0.99
I (314) efuse_init: Chip rev:         v0.2
I (319) heap_init: Initializing. RAM available for dynamic allocation:
I (327) heap_init: At 3FCA63A0 len 00043370 (268 KiB): RAM
I (333) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (339) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (345) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (352) spi_flash: detected chip: generic
I (356) spi_flash: flash io: qio
W (360) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (373) sleep: Configure to isolate all GPIO pins in sleep state
I (380) sleep: Enable automatic switching of GPIO sleep configuration
I (387) main_task: Started on CPU0
I (402) main_task: Calling app_main()
I (403) USBH_CDC: iot usbh cdc version: 2.0.0
I (433) usbh_rndis: IOT USBH RNDIS Version: 0.2.0
I (433) iot_eth: IoT ETH Version: 0.1.0
I (434) usbh_rndis: USB RNDIS network interface init success
I (439) iot_eth.netif_glue: 00:00:00:00:00:00
I (443) iot_eth.netif_glue: ethernet attached to netif
E (1449) usbh_rndis: RNDIS auto detect failed
W (1449) RNDIS_4G_MODULE: Failed to start USB RNDIS driver, try again...
E (3450) usbh_rndis: RNDIS auto detect failed
[... 283 lines cut ...]
I (37319) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
W (37361) phy_init: saving new calibration data because of checksum failure, mode(0)
I (37371) wifi:mode : softAP (30:ed:a0:ab:88:e5)
I (37372) wifi:Total power save buffer number: 32
I (37372) wifi:Init max length of beacon: 752/752
I (37373) wifi:Init max length of beacon: 752/752
I (37379) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
E (37387) ping_sock: send error=0
I (38296) iot_eth.netif_glue: Ethernet Got IP Address
I (38296) iot_eth.netif_glue: ~~~~~~~~~~~
I (38296) iot_eth.netif_glue: ETHIP:192.168.0.100
I (38301) iot_eth.netif_glue: ETHMASK:255.255.255.0
I (38306) iot_eth.netif_glue: ETHGW:192.168.0.1
I (38311) iot_eth.netif_glue: ~~~~~~~~~~~
W (39391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (44391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (49391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (54391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (59391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (64391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

yxmstar, 2025-06-13:
> 这个问题我也还没解决，sim卡data拉高，供电等都尝试过。卡也换过电信和移动的。

[](https://github.com/espressif/esp-iot-solution/issues/341#issuecomment-2965016920)
我现在情况和你一样，可以尝试一下是不是固件的问题

yxmstar, 2025-06-13:
> 这个问题我也还没解决，sim卡data拉高，供电等都尝试过。卡也换过电信和移动的。

我现在情况和你一样，你有试过直接接电脑可以用rndis上网吗

flycatac, 2025-06-14:
同样遇到过，不知道是不是同一个问题。

Ethernet Got IP Address 能获取到ip是因为用的路由模式不是网卡模式，实际上模块可能没有获取到ip。用AT指令`AT+MDIALUPCFG="workmode"[,<workmode>]`配置 0 是路由 1 是网卡

需要用AT 指令 `AT+CPIN?` `AT+CSQ` `AT+MDIALUP?` 来分别判断卡状态，网络信号，拨号状态。其中有一个有问题都无法上网。AT+MDIALUP有问题的话就要检查PDP上下文是否设置正确，特别是APN。

DPD配置AT:  `AT+CGDCONT=1,"IPV4V6","cmnet"` 
拨号AT: `AT+MDIALUP=1,1`

aiotfactory, 2025-12-24:
> 发件时间：2025年12月22日 16:50
>
>
> finally, it works, but it reports another error as below: I (268025) RNDIS_4G_MODULE: 64 bytes from 8.8.8.8 icmp_seq=1 ttl=113 time=69 ms
>
> I (273025) RNDIS_4G_MODULE: 64 bytes from 8.8.8.8 icmp_seq=1 ttl=113 time=69 ms
>
> I (278017) RNDIS_4G_MODULE: 64 bytes from 8.8.8.8 icmp_seq=1 ttl=113 time=61 ms
>
> I (278641) wifi:new:<1,1&gt;, old:<1,1&gt;, ap:<1,1&gt;, sta:<255,255&gt;, prof:1, snd_ch_cfg:0x0 I (278642) wifi:station: 02:d4:34:eb:61:f3 join, AID=1, bgn, 40U I (278644) app wifi: station 02:d4:34:eb:61:f3 join, AID=1 I (278650) app wifi: ip_napt_enable I (278814) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2 I (280320) wifi:idx:2 (ifx:1, 02:d4:34:eb:61:f3), tid:0, ssn:10, winSize:64 E (280402) usbh_rndis: Error rndis packet message E (280402) usbh_rndis: Error rndis packet message E (280404) usbh_rndis: Error rndis packet message E (280407) usbh_rndis: Error rndis packet message E (280413) usbh_rndis: Error rndis packet message E (280484) usbh_rndis: Error rndis packet message E (280732) usbh_rndis: Error rndis packet message E (280732) usbh_rndis: Error rndis packet message E (281204) usbh_rndis: Error rndis packet message E (281204) usbh_rndis: Error rndis packet message E (282127) usbh_rndis: Error rndis packet message E (282127) usbh_rndis: Error rndis packet message E (282471) usbh_rndis: Error rndis packet message E (282471) usbh_rndis: Error rndis packet message E (282473) usbh_rndis: Error rndis packet message E (282476) usbh_rndis: Error rndis packet message E (282567) usbh_rndis: Error rndis packet message E (282567) usbh_rndis: Error rndis packet message E (282831) usbh_rndis: Error rndis packet message E (2828 [... cut ...] 54 ms
>
> E (283390) usbh_rndis: Error rndis packet message E (283390) usbh_rndis: Error rndis packet message E (284045) usbh_rndis: Error rndis packet message E (284046) usbh_rndis: Error rndis packet message E (284478) usbh_rndis: Error rndis packet message
>
> 这个问题应该和 #619 是一样的，是数据出现了粘包导致的，我们将很快修复这个问题。

期待你的修复，目前很麻烦，有快速解决办法么，多谢。

Peter
***@***.***

        

         原始邮件

aiotfactory, 2025-12-25:
或者有什么其他临时办法么，比如如何配置较小包。多谢。

  

 
 
 
 
  
这个问题应该和 #619 是一样的，是数据出现了粘包导致的，我们将很快修复这个问题。