espressif/esp-iot-solution #518: iot_button_create error when building in Version 4.1.3

andresragot, 2025-05-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Steps to reproduce.

I added the component using the add-dependency command:

`idf.py add-dependency "espressif/button=*"`

This downloaded the latest version, which is 4.1.3.

I am currently encapsulating this C API into a C++ class.

In the constructor, I simply implemented the example provided in the documentation.

```cpp
...
button_config_t gpio_btn_cfg = {
            .type = BUTTON_TYPE_GPIO,
            .long_press_time = CONFIG_BUTTON_LONG_PRESS_TIME_MS,
            .short_press_time = CONFIG_BUTTON_SHORT_PRESS_TIME_MS,
            .gpio_button_config = {
                .gpio_num = _pin,
                .active_level = 0,
            },
        };

        _btn_handle = iot_button_create(&gpio_btn_cfg);
...
```

However, I get an error when building the app with idf.py build:

```
/Driver_Button.cpp:35:40: error: too few arguments to function 'esp_err_t iot_button_create(const button_config_t*, const button_driver_t*, button_dev_t**)'
   35 |         _btn_handle = iot_button_create(&gpio_btn_cfg);
      |                       ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
```

When I checked the source files, I found that the signature of the `iot_button_create` function had changed. It is now:

```cpp
esp_err_t iot_button_create(const button_config_t *config, const button_driver_t *driver, button_handle_t *ret_button);
```

So I though, "Okay, no problem, let's keep going based on what I can see in the files" and I updated my code as follows:

```cpp
...
 button_config_t gpio_btn_cfg = {
            .long_press_time = CONFIG_BUTTON_LONG_PRESS_TIME_MS,
            .short_press_time = CONFIG_BUTTON_SHORT_PRESS_TIME_MS,
        };

        button_driver_t gpio_btn_driver = {
            .enable_power_save = false,  // Power saving is not enabled by default
            .get_key_level = [] (button_driver_t * driver) -> uint8_t {
                return 0; // Default to not pressed
            },
            .enter_power_save = nullptr, // No power save function
            .del = nullptr, // No delete function
        };

        button_dev_t gpio_btn_dev = {
            .long_press_hold_cnt = config.holdTimeMs,
            .repeat = 0,
            .driver = &gpio_btn_driver,
            .event = BUTTON_NONE_PRESS
        };

        _btn_handle = iot_button_create(&gpio_btn_cfg, &gpio_btn_driver, &gpio_btn_dev);
...
```
But the I got the following error:

```
/Driver_Button.cpp:37:22: error: variable 'button_dev_t gpio_btn_dev' has initializer but incomplete type
   37 |         button_dev_t gpio_btn_dev = {
      |                      ^~~~~~~~~~~~
```
This is because the `button_dev_t` struct is defined inside the `.c` file and there is no forward declaration or public header exposing it.

### Build Logs.

```plain
/Driver_Button.cpp:35:40: error: too few arguments to function 'esp_err_t iot_button_create(const button_config_t*, const button_driver_t*, button_dev_t**)'
   35 |         _btn_handle = iot_button_create(&gpio_btn_cfg);
      |                       ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
```

### More Information.

The examples only work if I downgrade the component to version 3.5.0.

lijunru-hub, 2025-05-30:
This update is a major version upgrade. If you prefer not to modify the API interface, please restrict the component to a version below 4.

andresragot, 2025-05-30:
Thanks. I couldn’t locate the initialization method, and it looks like the documentation hasn’t been updated in a while to cover the new features introduced in the latest major release.

lijunru-hub, 2025-06-03:
Sorry, some parts of the English documentation for the IoT button were missing and have now been updated. In the meantime, please refer to the code examples provided in the Chinese documentation.

closed 2025-06-03 by andresragot