espressif/esp-iot-solution #520: ESP32S3 UVC 包大小一直返回500

ffalsk, 2025-05-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### 环境
ESP-IDF 5.4
ESP32S3N16R8
使用示例usb_camera_mic_spk

### 问题
通过浏览器查看没有图像显示，包大小一直显示500

### 日志
以下是USB相关日志
```
I (826) USB_STREAM: USB stream task start
D (856) USB_STREAM: Port=1 init succeed, context = 0x3fcc4d80
I (856) USB_STREAM: USB Streaming Start Succeed
D (856) USB_STREAM: USB Task: Waiting Connection
I (856) USB_STREAM: Waiting USB Device Connection
I (1107) USB_STREAM: line 151 HCD_PORT_EVENT_CONNECTION
D (1107) USB_STREAM: port update: action_bits = 0x0008
I (1108) USB_STREAM: Action: ACTION_DEVICE_CONNECT
I (1158) USB_STREAM: Resetting Port
I (1218) USB_STREAM: Setting Port FIFO, 1
D (1218) USB_STREAM: Action: ACTION_DEVICE_CONNECT, Done!
D (1218) USB_STREAM: Action: ACTION_DEVICE_ENUM Stage NONE (0)
I (1219) USB_STREAM: USB Speed: full-speed
D (1219) USB_STREAM: pipe(0x3fcc4ed4) init succeed
I (1219) USB_STREAM: ENUM Stage START, Succeed
D (1219) USB_STREAM: Action: ACTION_DEVICE_ENUM, Done (0)
D (1220) USB_STREAM: Action: ACTION_DEVICE_ENUM Stage GET_SHORT_DEV_DESC (2)
I (1220) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
D (1221) USB_STREAM: Action: ACTION_DEVICE_ENUM, Waiting URB Done
D (1221) USB_STREAM: Action: ACTION_DEVICE_ENUM, Done (2)
V (1221) USB_STREAM: Pipe(default): XFER_DONE
D (1221) USB_STREAM: pipe update: action_bits = 0x4000
D (1221) USB_STREAM: Action: ACTION_PIPE_XFER_DONE
D (1222) USB_STREAM: Action: ACTION_PIPE_XFER_DONE, Done!
D (1223) USB_STREAM: Action: ACTION_DEVICE_ENUM Stage CHECK_SHORT_DEV_DESC (3)
I (1223) USB_STREAM: Default pipe endpoint MPS update to 64
I (1223) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Succeed
D (1224) USB_STREAM: Action: ACTION_DEVICE_ENUM, Done (3)
D (1225) USB_STREAM: Action: ACTION_DEVICE_ENUM Stage SET_ADDR (4)
I (1225) USB_STREAM: ENUM Stage SET_ADDR, ucceed
D (1225) USB_STREAM: Action: ACTION_DEVICE_ENUM, Waiting URB Done
D (1225) USB_STREAM: Action: ACTION_DEVICE_ENUM, Done (4)
V (1226) USB_STREAM: Pipe(default): XFER_DONE
D (1226) USB_STREAM: pipe update: action_bits = 0x4000
D (1226) USB_STREAM: Action: ACTION_PIPE_XFER_DONE
D (1226) USB_STREAM: Action: ACTION_PIPE_XFER_DONE, Done!
D (1228) USB_STREAM: Action: ACTION_DEVICE_ENUM Stage CHECK_ADDR (5)
I (1238) USB_STREAM: ENUM Stage CHECK_ADDR, Succeed
D (1238) USB_STREAM: Action: ACTION_DEVICE_ENUM, Done (5)
D (1239) USB_STREAM: Action: ACTION_DEVICE_ENUM Stage GET_FULL_DEV_DESC (6)
I (1239) USB_STREAM: ENUM Stage GET_FULL_DEV_DESC, Succeed
D (1239) USB_STREAM: Action: ACTION_DEVICE_ENUM, Waiting URB Done
D (1240) USB_STREAM: Action: ACTION_DEVICE_ENUM, Done (6)
V (1240) USB_STREAM: Pipe(default): XFER_DONE
D (1240) USB_STREAM: pipe update: action_bits = 0x4000
D (1240) USB_STREAM: Action: ACTION_PIPE_XFER_DONE
D (1241) USB_STREAM: Action: ACTION_PIPE_XFER_DONE, Done!
D (1242) USB_STREAM: Action: ACTION_DEVICE_ENUM Stage CHECK_FULL_DEV_DESC (7)
*** Device descriptor ***
bcdUSB 2.00
bDeviceClass 0xef
[... 371 lines cut ...]
D (42769) USBSTREAM: bogus packet: len = 512eb b1...51 45

V (42769) USB_STREAM: Pipe(UVC): XFER_DONE
V (42769) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x02, second=0x16
D (42770) USB_STREAM: bogus packet: len = 512 02 16...fc e6

V (42770) USB_STREAM: Pipe(UVC): XFER_DONE
V (42770) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0xaa, second=0x9c
D (42771) USB_STREAM: bogus packet: len = 512 aa 9c...62 51

V (42771) USB_STREAM: Pipe(UVC): XFER_DONE
V (42771) USB_STREAM: zlp=0, lstp=1, req_len=512, payload_len=287, first=0x4e, second=0xff
D (42772) USB_STREAM: bogus packet: len = 287 4e ff...ff d9

V (42808) USB_STREAM: Pipe(UVC): XFER_DONE
V (42808) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x0c, second=0x8d
V (42808) USB_STREAM: header=12 info=0x8d, payload_len = 512, last_pts = 11790 , last_scr = 11790
D (42809) USB_STREAM: SWAP NO EOF 500
V (42809) USB_STREAM: uvc swap buffer length = 500
V (42809) USB_STREAM: uvc payloa = ff d8...2a 35

V (42809) USB_STREAM: GOT LENGTH 500
V (42810) USB_STREAM: Pipe(UVC): XFER_DONE
I (42810) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 187, width = 480, height = 320, length = 500, ptr = 0
```

ffalsk, 2025-05-30:
以下是windows抓取的USB信息
```

    =========================== USB Port2 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 3-1-1-2
Properties               : 0x00
 IsUserConnectable       : no
 PortIsDebugCapable      : no
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x02 (Port 2)

      ========================== Summary =========================
Vendor ID                : 0x09A1 (ELMO COMPANY, LIMITED)
Product ID               : 0x002A
Manufacturer String      : SZFugesys
Product String           : HD CAM
Serial                   : 20250215
USB Version              : 2.0
Port maximum Speed       : High-Speed
Device maximum Speed     : High-Speed
Device Connection Speed  : Full-Speed
Self powered             : no
Demanded Current         : 500 mA
Used Endpoints           : 6
Video                    : UVC Version 1.00
                 -------- Default Video Modes --------
1280 x 720 @ 30 fps      : MJPEG
1280 x 720 @ 15 fps      : MJPEG
960 x 540 @ 30 fps       : MJPEG
960 x 540 @ 15 fps       : MJPEG
864 x 480 @ 30 fps       : MJPEG
864 x 480 @ 15 fps       : MJPEG
800 x 480 @ 30 fps       : MJPEG
800 x 480 @ 15 fps       : MJPEG
480 x 640 @ 30 fps       : MJPEG
480 x 640 @ 15 fps       : MJPEG
480 x 320 @ 30 fps       : MJPEG
320 x 480 @ 30 fps       : MJPEG
480 x 320 @ 15 fps       : MJPEG
320 x 480 @ 15 fps       : MJPEG
320 x 240 @ 30 fps       : MJPEG
320 x 240 @ 15 fps       : MJPEG

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\?\USB#VID_09A1&PID_002A#20250215#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-16
Device ID                : USB\VID_09A1&PID_002A\20250215
Hardware IDs             : USB\VID_09A1&PID_002A&REV_0409 USB\VID_09A1&PID_002A
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0024 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.26100.3912  Date: 2025-05-14  Company: Microsoft Corporation)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0002.Hub_#0008
Address                  : 2
Location IDs             : PCIROOT(0)#PCI(0803)#PCI(0004)#USBROOT(0)#USB(1)#USB(1)#USB(2), ACPI(_SB_)#ACPI(PCI0)#ACPI(GP19)#ACPI(XHC4)#ACPI(RHUB)#ACPI(PRT1)#USB(1)#USB(2)
Container ID             : {526e8215-2fc3-555e-b98c-d1d1381c6536}
Manufacturer Info        : (标准 USB 主控制器)
Capabilities             : 0x94 (Removable, UniqueID, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
[... 1118 lines cut ...]
Language 0x0409          : "Video Streaming"
Data (HexDump)           : 20 03 56 00 69 00 64 00 65 00 6F 00 20 00 53 00    .V.i.d.e.o. .S.
                           74 00 72 00 65 00 61 00 6D 00 69 00 6E 00 67 00   t.r.e.a.m.i.n.g.
             ------ String Descriptor 8 ------
bLength                  : 0x42 (66 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "CDC Abstract Control Model (ACM)"
Data (HexDump)           : 42 03 43 00 44 00 43 00 20 00 41 00 62 00 73 00   B.C.D.C. .A.b.s.
                           74 00 72 00 61 00 63 00 74 00 20 00 43 00 6F 00   t.r.a.c.t. .C.o.
                           6E 00 74 00 72 00 6F 00 6C 00 20 00 4D 00 6F 00   n.t.r.o.l. .M.o.
                           64 00 65 00 6C 00 20 00 28 00 41 00 43 00 4D 00   d.e.l. .(.A.C.M.
                           29 00                                             ).
             ------ String Descriptor 9 ------
bLength                  : 0x1A (26 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "CDC ACM Data"
Data (HexDump)           : 1A 03 43 00 44 00 43 00 20 00 41 00 43 00 4D 00   ..C.D.C. .A.C.M.
                           20 00 44 00 61 00 74 00 61 00                      .D.a.t.a.
             ------ String Descriptor 10 ------
bLength                  : 0x16 (22 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "CDC Serial"
Data (HexDump)           : 16 03 43 00 44 00 43 00 20 00 53 00 65 00 72 00   ..C.D.C. .S.e.r.
                           69 00 61 00 6C 00                                 i.a.l.
```

ffalsk, 2025-05-30:
@leeebo 

**没有任何显示**

[image]

```
I (534170) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 2226, width = 320, height = 240, length = 500, ptr = 0
V (534170) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x36, second=0x37
D (534171) USB_STREAM: bogus packet: len = 512 36 37...6a b1

D (534171) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb3884
V (534172) uvc_mic_spk_demo: send frame = 2226
D (534172) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb3884
V (534171) USB_STREAM: Pipe(UVC): XFER_DONE
V (534173) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x2b, second=0x77
D (534174) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb3884
I (534174) camera_httpd: MJPG: 500B 50ms (20.0fps), AVG: 54ms (18.5fps)
V (534174) uvc_mic_spk_demo: send frame done = 2226
D (534174) USB_STREAM: bogus packet: len = 512 2b 77...06 4b

V (534175) USB_STREAM: Pipe(UVC): XFER_DONE
V (534175) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x45, second=0x30
D (534176) USB_STREAM: bogus packet: len = 512 45 30...19 2a

V (534176) USB_STREAM: Pipe(UVC): XFER_DONE
V (534176) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0xb1, second=0xc5
D (534177) USB_STREAM: bogus packet: len = 512 b1 c5...52 d1

V (534177) USB_STREAM: Pipe(UVC): XFER_DONE
V (534177) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x45, second=0x05
D (534178) USB_STREAM: bogus packet: len = 512 45 05...a7 d1

V (534178) USB_STREAM: Pipe(UVC): XFER_DONE
V (534178) USB_STREAM: zlp=0, lstp=1, req_len=512, payload_len=478, first=0x54, second=0x3e
D (534179) USB_STREAM: bogus packet: len = 478 54 3e...ff d9

V (534228) USB_STREAM: Pipe(UVC): XFER_DONE
V (534228) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x0c, second=0x8c
V (534229) USB_STREAM: header=12 info=0x8c, payload_len = 512, last_pts = 126480 , last_scr = 126480
D (534229) USB_STREAM: SWAP NO EOF 500
V (534229) USB_STREAM: uvc swap buffer length = 500
V (534229) USB_STREAM: uvc payload = ff d8...2a 35

V (534229) USB_STREAM: GOT LENGTH 500
V (534230) USB_STREAM: Pipe(UVC): XFER_DONE
I (534230) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 2227, width = 320, height = 240, length = 500, ptr = 0
V (534230) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x36, second=0x37
D (534231) USB_STREAM: bogus packet: len = 512 36 37...d3 a1

D (534231) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb3884
V (534232) uvc_mic_spk_demo: send frame = 2227
D (534232) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb3884
V (534232) USB_STREAM: Pipe(UVC): XFER_DONE
V (534233) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x97, second=0x25
D (534233) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb3884
I (534233) camera_httpd: MJPG: 500B 59ms (16.9fps), AVG: 54ms (18.5fps)
D (534233) USB_STREAM: bogus packet: len = 512 97 25...1f e7

V (534233) uvc_mic_spk_demo: send frame done = 2227
V (534235) USB_STREAM: Pipe(UVC): XFER_DONE
V (534235) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512 first=0xfc, second=0xff
D (534236) USB_STREAM: bogus packet: len = 512 fc ff...64 71

V (534236) USB_STREAM: Pipe(UVC): XFER_DONE
V (534236) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0xb2, second=0x3a
D (534237) USB_STREAM: bogus packet: len = 512 b2 3a...f9 7a

V (534237) USB_STREAM: Pipe(UVC): XFER_DONE
V (534237) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x8e, second=0xbd
D (534238) USB_STREAM: bogus packet: len = 512 8e bd...0e 6b

V (534238) USB_STREAM: Pipe(UVC): XFER_DONE
V (53238) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x11, second=0x52
D (534239) USB_STREAM: bogus packet: len = 512 11 52...00 f5

V (534239) USB_STREAM: Pipe(UVC): XFER_DONE
V (534239) USB_STREAM: zlp=0, lstp=1, req_len=512, payload_len=28, first=0xfe, second=0xb5
D (534240) USB_STREAM: bogus packet: len = 28 fe b5...ff d9

V (534278) USB_STREAM: Pipe(UVC): XFER_DONE
V (534278) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x0c, second=0x8d
V (534278) USB_STREAM: header=12 info=0x8d, payload_len = 512, last_pts = 126540 , last_scr = 126540
D (534279) USB_STREAM: SWAP NO EOF 500
V (534279) USB_STREAM: uvc swap buffer length = 500
V (534279) USB_STREAM: uvc payload = ff d8...2a 35

V (534279) USB_STREAM: GOT LENGTH 500
V (534279) USB_STREAM: Pipe(UVC): XFER_DONE
I (534280) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 2228, width = 320, height = 240, length = 500, ptr = 0
V (534280) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x36, second=0x37
D (534281) USB_STREAM: bogus packet: len = 512 36 37...df fc

D (534281) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb3884
V (534282) uvc_mic_spk_demo: send frame = 2228
D (534282) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb3884
V (534281) USB_STREAM: Pipe(UVC): XFER_DONE
V (534283) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0x2a, second=0x9b
D (534284) USB_STREAM: bogus packet: len = 512 2a 9b...28 a0

D (534284) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb3884
V (534284) USB_STREAM: Pipe(UVC): XFER_DONE
I (534284) camera_httpd: MJPG: 500B 50ms (20.0fps), AVG: 54ms (18.5fps)
V (534285) USB_STREAM: zlp=0, lstp=0, req_len=512, payload_len=512, first=0xb2, second=0x4a
V (534284) uvc_mic_spk_demo: send frame done = 2228
D (534287) USB_STREAM: bogus packet: len = 512 b2 4a...11 7f
```

ffalsk, 2025-05-31:
经过分析，发现ESP32协商到了错误的`usbvideo.probe.maxPayloadTransferSize`，在windows上是768000，而esp32得到了512，导致后续esp32的驱动认为包大小不会超过512。
目前强制指定`usbvideo.probe.maxPayloadTransferSize`后工作正常。
具体为
``` c++
ctrl_probed.dwMaxPayloadTransferSize = 768000;
```

[image]

leeebo, 2025-06-03:
@ffalsk  如果同在 Fullspeed 下（Windows 按照 Fullspeed 枚举设备），协商结果不应该有差异。

是否可以连接摄像头供应商，咨询什么条件下会协商到 `dwMaxPayloadTransferSize=512`