espressif/esp-iot-solution #521: bh1750 fails to build when CONFIG_I2C_BUS_BACKWARD_CONFIG is not set

ozanoner, 2025-05-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-C6FH4 (QFN32) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

https://www.waveshare.com/wiki/ESP32-C6-Zero

### Steps to reproduce.

1. Step: Extract the attached example project (CONFIG_I2C_BUS_BACKWARD_CONFIG is not set, which is the default)
2. Step: Clone the master repo (commit 6f25eba1c26d6220cdf28ae3a13532a31c22d688) in the project directory
3. Step: run `idf.py build`

[multis_ex.tar.gz](https://github.com/user-attachments/files/20532916/multis_ex.tar.gz)

### Build Logs.

```plain
$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/ozan/projects/tmp/multis_ex/build
Executing "ninja all"...
[7/31] Building C object esp-idf/bh1750/CMakeFiles/__idf_bh1750.dir/bh1750.c.obj
FAILED: esp-idf/bh1750/CMakeFiles/__idf_bh1750.dir/bh1750.c.obj 
/home/ozan/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.4\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/ozan/projects/tmp/multis_ex/build/config -I/home/ozan/projects/tmp/multis_ex/esp-iot-solution/components/sensors/light_sensor/bh1750/include -I/home/ozan/esp/esp-idf/components/newlib/platform_include -I/home/ozan/esp/esp-idf/components/freertos/config/include -I/home/ozan/esp/esp-idf/components/freertos/config/include/freertos -I/home/ozan/esp/esp-idf/components/freertos/config/riscv/include -I/home/ozan/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/ozan/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/ozan/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I/home/ozan/esp/esp-idf/components/freertos/esp_additions/include -I/home/ozan/esp/esp-idf/components/esp_hw_support/include -I/home/ozan/esp/esp-idf/components/esp_hw_support/include/soc -I/home/ozan/esp/esp-idf/components/esp_hw_support/include/soc/esp32c6 -I/home/ozan/esp/esp-idf/components/esp_hw_support/dma/include -I/home/ozan/esp/esp-idf/components/esp_hw_support/ldo/include -I/home/ozan/esp/esp-idf/components/esp_hw_support/debug_probe/include -I/home/ozan/esp/esp-idf/components/esp_hw_support/port/esp32c6/. -I/home/ozan/esp/esp-idf/components/esp_hw_support/port/esp32c6/include -I/home/ozan/esp/esp-idf/components/esp_hw_support/port/esp32c6/private_include -I/home/ozan/esp/esp-idf/components/heap/include -I/home/ozan/esp/esp-idf/components/heap/tlsf -I/home/ozan/esp/esp-idf/components/log/include -I/home/ozan/esp/esp-idf/components/soc/include -I/home/ozan/esp/esp-idf/components/soc/esp32c6 -I/home/ozan/esp/esp-idf/components/soc/esp32c6/include -I/home/ozan/esp/esp-idf/components/soc/esp32c6/register -I/home/ozan/esp/esp-idf/components/hal/platform_port/include -I/home/ozan/esp/esp-idf/components/hal/esp32c6/include -I/home/ozan/esp/esp-idf/components/hal/include -I/home/ozan/esp/esp-idf/components/esp_rom/include -I/home/ozan/esp/esp-idf/components/esp_rom/esp32c6/include -I/home/ozan/esp/esp-idf/components/esp_rom/esp32c6/include/esp32c6 -I/home/ozan/esp/esp-idf/components/esp_rom/esp32c6 -I/home/ozan/esp/esp-idf/components/esp_common/include -I/home/ozan/esp/esp-idf/components/esp_system/include -I/home/ozan/esp/esp-idf/components/esp_system/port/soc -I/home/ozan/esp/esp-idf/components/esp_system/port/include/riscv -I/home/ozan/esp/esp-idf/components/esp_system/port/include/private -I/home/ozan/esp/esp-idf/components/riscv/include -I/home/ozan/esp/esp-idf/components/lwip/include -I/home/ozan/esp/esp-idf/components/lwip/include/apps -I/home/ozan/esp/esp-idf/components/lwip/include/apps/sntp -I/home/ozan/esp/esp-idf/components/lwip/lwip/src/include -I/home/ozan/esp/esp-idf/components/lwip/port/include -I/home/ozan/esp/esp-idf/components/lwip/port/freertos/include -I/home/ozan/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/ozan/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/ozan/esp/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/ozan/projects/tmp/multis_ex/esp-iot-solution/components/i2c_bus/include -I/home/ozan/esp/esp-idf/components/esp_driver_i2c/include -I/home/ozan/esp/esp-idf/components/driver/deprecated -I/home/ozan/esp/esp-idf/components/driver/i2c/include -I/home/ozan/esp/esp-idf/components/driver/touch_sensor/include -I/home/ozan/esp/esp-idf/components/driver/twai/include -I/home/ozan/esp/esp-idf/components/esp_pm/include -I/home/ozan/esp/esp-idf/components/esp_ringbuf/include -I/home/ozan/esp/esp-idf/components/esp_driver_gpio/include -I/home/ozan/esp/esp-idf/components/esp_driver_pcnt/include -I/home/ozan/esp/esp-idf/components/esp_driver_gptimer/include -I/home/ozan/esp/esp-idf/components/esp_driver_spi/include -I/home/ozan/esp/esp-idf/components/esp_driver_mcpwm/include -I/home/ozan/esp/esp-idf/components/esp_driver_ana_cmpr/include -I/home/ozan/esp/esp-idf/components/esp_driver_i2s/include -I/home/ozan/esp/esp-idf/components/esp_driver_sdmmc/include -I/home/ozan/esp/esp-idf/components/sdmmc/include -I/home/ozan/esp/esp-idf/components/esp_driver_sdspi/include -I/home/ozan/esp/esp-idf/components/esp_driver_sdio/include -I/home/ozan/esp/esp-idf/components/esp_driver_dac/include -I/home/ozan/esp/esp-idf/components/esp_driver_rmt/include -I/home/ozan/esp/esp-idf/components/esp_driver_tsens/include -I/home/ozan/esp/esp-idf/components/esp_driver_sdm/include -I/home/ozan/esp/esp-idf/components/esp_driver_uart/include -I/home/ozan/esp/esp-idf/components/vfs/include -I/home/ozan/esp/esp-idf/components/esp_driver_ledc/include -I/home/ozan/esp/esp-idf/components/esp_driver_parlio/include -I/home/ozan/esp/esp-idf/components/esp_driver_usb_serial_jtag/include -march=rv32imac_zicsr_zifencei  -fdiagnostics-color=always -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Og -fno-shrink-wrap -fmacro-prefix-map=/home/ozan/projects/tmp/multis_ex=. -fmacro-prefix-map=/home/ozan/esp/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -DI2C_BUS_VER_MAJOR=1 -DI2C_BUS_VER_MINOR=4 -DI2C_BUS_VER_PATCH=0 -MD -MT esp-idf/bh1750/CMakeFiles/__idf_bh1750.dir/bh1750.c.obj -MF esp-idf/bh1750/CMakeFiles/__idf_bh1750.dir/bh1750.c.obj.d -o esp-idf/bh1750/CMakeFiles/__idf_bh1750.dir/bh1750.c.obj -c /home/ozan/projects/tmp/multis_ex/esp-iot-solution/components/sensors/light_sensor/bh1750/bh1750.c
/home/ozan/projects/tmp/multis_ex/esp-iot-solution/components/sensors/light_sensor/bh1750/bh1750.c: In function 'bh1750_get_light_intensity':
/home/ozan/projects/tmp/multis_ex/esp-iot-solution/components/sensors/light_sensor/bh1750/bh1750.c:110:9: error: implicit declaration of function 'vTaskDelay' [-Wimplicit-function-declaration]
  110 |         vTaskDelay(30 / portTICK_RATE_MS);
      |         ^~~~~~~~~~
In file included from /home/ozan/projects/tmp/multis_ex/esp-iot-solution/components/sensors/light_sensor/bh1750/include/bh1750.h:10,
                 from /home/ozan/projects/tmp/multis_ex/esp-iot-solution/components/sensors/light_sensor/bh1750/bh1750.c:8:
/home/ozan/projects/tmp/multis_ex/esp-iot-solution/components/i2c_bus/include/i2c_bus.h:54:26: error: 'portTICK_PERIOD_MS' undeclared (first use in this function); did you mean 'portTICK_RATE_MS'?
   54 | #define portTICK_RATE_MS portTICK_PERIOD_MS
      |                          ^~~~~~~~~~~~~~~~~~
/home/ozan/projects/tmp/multis_ex/esp-iot-solution/components/sensors/light_sensor/bh1750/bh1750.c:110:25: note: in expansion of macro 'portTICK_RATE_MS'
  110 |         vTaskDelay(30 / portTICK_RATE_MS);
[... cut ...]
ninja failed with exit code 1, output of the command is in the /home/ozan/projects/tmp/multis_ex/build/log/idf_py_stderr_output_566186 and /home/ozan/projects/tmp/multis_ex/build/log/idf_py_stdout_output_566186
```

### More Information.

When CONFIG_I2C_BUS_BACKWARD_CONFIG is set (=y), it builds successfully.

Potential solution: When I update the esp-iot-solution/components/sensors/light_sensor/bh1750/bh1750.c file with the FreeRTOS task header, it builds successfully whether CONFIG_I2C_BUS_BACKWARD_CONFIG is set or not.

YanKE01, 2025-06-03:
Hi, @ozanoner . Thank you for your attention to the `BH1750` component. The issue you mentioned is mainly due to the use of RTOS functions in the `BH1750` component without including the necessary RTOS header files. This issue will be fixed soon.

ozanoner, 2025-06-04:
> in the `BH1750` component without including the necessary RTOS header files. 

Actually, this was another thing I wanted to ask. It is not a component yet (no idf_component.yml in the root of BH1750)

https://github.com/espressif/esp-iot-solution/tree/master/components/sensors/light_sensor/bh1750

There is a component on ESP-component-registry here: https://components.espressif.com/components/espressif/bh1750/versions/1.0.3 

But I think that one is independent from this repository.

YanKE01, 2025-06-05:
> Actually, this was another thing I wanted to ask. It is not a component yet (no idf_component.yml in the root of BH1750) [...]

Just as you said, the [BH1750](https://components.espressif.com/components/espressif/bh1750/versions/1.0.3) component is maintained by the ESP-BSP team. The BH1750 in esp-iot-solution does not exist in the current ESP Registry.