espressif/esp-iot-solution #549: usb_stream 1.5.1 build failed on idf5.4.1

yunyizhi, 2025-07-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf5.4.1

### Espressif SoC revision.

 ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32 s3 devkit-c

### Steps to reproduce.

* Set the target to esp32s3
* Import the component `espressif/usb_stream: ^1.5.1`
* The following code is used in the project
```c
/*
 * SPDX-FileCopyrightText: 2022-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include "usb_camera.h"
#include <assert.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "esp_err.h"
#include "esp_log.h"
#include "usb_stream.h"

static const char *TAG = "uvc_demo";

#define BIT0_FRAME_START     (0x01 << 0)
#define BIT1_NEW_FRAME_START (0x01 << 1)
#define BIT2_NEW_FRAME_END   (0x01 << 2)

#define DEMO_UVC_FRAME_WIDTH        480
#define DEMO_UVC_FRAME_HEIGHT       320

#ifdef CONFIG_IDF_TARGET_ESP32S2
#define DEMO_UVC_XFER_BUFFER_SIZE (45 * 1024)
#else
#define DEMO_UVC_XFER_BUFFER_SIZE (55 * 1024)
#endif

#define BIT0_FRAME_START     (0x01 << 0)

static void stream_state_changed_cb(usb_stream_state_t event, void *arg) {
    switch (event) {
        case STREAM_CONNECTED: {
            size_t frame_size = 0;
            size_t frame_index = 0;
            uvc_frame_size_list_get(NULL, &frame_size, &frame_index);
            if (frame_size) {
                ESP_LOGI(TAG, "UVC: get frame list size = %u, current = %u", frame_size, frame_index);
                uvc_frame_size_t *uvc_frame_list = (uvc_frame_size_t *) malloc(frame_size * sizeof(uvc_frame_size_t));
                uvc_frame_size_list_get(uvc_frame_list, NULL, NULL);
                for (size_t i = 0; i < frame_size; i++) {
                    ESP_LOGI(TAG, "\tframe[%u] = %ux%u", i, uvc_frame_list[i].width, uvc_frame_list[i].height);
                }
                free(uvc_frame_list);
            } else {
                ESP_LOGW(TAG, "UVC: get frame list size = %u", frame_size);
            }
            ESP_LOGI(TAG, "Device connected");
            break;
        }
        case STREAM_DISCONNECTED:
            ESP_LOGI(TAG, "Device disconnected");
            break;
        default:
            ESP_LOGE(TAG, "Unknown event");
            break;
    }
}

void usb_camera_start() {
    esp_err_t ret = ESP_FAIL;
    /* malloc double buffer for usb payload, xfer_buffer_size >= frame_buffer_size*/
    uint8_t *xfer_buffer_a = (uint8_t *) heap_caps_malloc(DEMO_UVC_XFER_BUFFER_SIZE,
                                                          MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM);
    assert(xfer_buffer_a != NULL);
    uint8_t *xfer_buffer_b = (uint8_t *) heap_caps_malloc(DEMO_UVC_XFER_BUFFER_SIZE,
                                                          MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM);
    assert(xfer_buffer_b != NULL);

    /* malloc frame buffer for a jpeg frame*/
    uint8_t *frame_buffer = (uint8_t *) heap_caps_malloc(DEMO_UVC_XFER_BUFFER_SIZE,
                                                         MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM);
    assert(frame_buffer != NULL);

    uvc_config_t uvc_config = {
            /* match the any resolution of current camera (first frame size as default) */
            .frame_width = DEMO_UVC_FRAME_WIDTH,
            .frame_height = DEMO_UVC_FRAME_HEIGHT,
            .frame_interval = FRAME_INTERVAL_FPS_15,
            .xfer_buffer_size = DEMO_UVC_XFER_BUFFER_SIZE,
            .xfer_buffer_a = xfer_buffer_a,
            .xfer_buffer_b = xfer_buffer_b,
            .frame_buffer_size = DEMO_UVC_XFER_BUFFER_SIZE,
            .frame_buffer = frame_buffer,
            .frame_cb = &usb_camera_frame_cb,
            .frame_cb_arg = NULL,
    };
    /* config to enable uvc function */
    ret = uvc_streaming_config(&uvc_config);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "uvc streaming config failed");
    }
    /* register the state callback to get connect/disconnect event
    * in the callback, we can get the frame list of current device
    */
    ESP_ERROR_CHECK(usb_streaming_state_register(&stream_state_changed_cb, NULL));
    /* start usb streaming, UVC and UAC MIC will start streaming because SUSPEND_AFTER_START flags not set */
    ESP_ERROR_CHECK(usb_streaming_start());
    ESP_ERROR_CHECK(usb_streaming_connect_wait(portMAX_DELAY));
}

```
* Run `idf.py build`

### Build Logs.

```plain
D:/Users/immor/CLionProjects/espressif-demo/cam/dual_cam_idf/managed_components/espressif__usb_stream/usb_host_helpers.c: In function '_usb_port_init':
D:/Users/immor/CLionProjects/espressif-demo/cam/dual_cam_idf/managed_components/espressif__usb_stream/usb_host_helpers.c:200:10: error: 'hcd_config_t' has no member named 'peripheral_map'
  200 |         .peripheral_map = BIT0,
      |          ^~~~~~~~~~~~~~
In file included from D:/Users/immor/CLionProjects/espressif-demo/cam/dual_cam_idf/managed_components/espressif__usb_stream/usb_host_helpers.c:17:
D:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_common/include/esp_bit_defs.h:41:18: warning: excess elements in struct initializer
   41 | #define BIT0     0x00000001
      |                  ^~~~~~~~~~
D:/Users/immor/CLionProjects/espressif-demo/cam/dual_cam_idf/managed_components/espressif__usb_stream/usb_host_helpers.c:200:27: note: in expansion of macro 'BIT0'
  200 |         .peripheral_map = BIT0,
      |                           ^~~~
D:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_common/include/esp_bit_defs.h:41:18: note: (near initialization for 'hcd_config')
   41 | #define BIT0     0x00000001
      |                  ^~~~~~~~~~
D:/Users/immor/CLionProjects/espressif-demo/cam/dual_cam_idf/managed_components/espressif__usb_stream/usb_host_helpers.c:200:27: note: in expansion of macro 'BIT0'
  200 |         .peripheral_map = BIT0,
      |                           ^~~~
D:/Users/immor/CLionProjects/espressif-demo/cam/dual_cam_idf/managed_components/espressif__usb_stream/usb_host_helpers.c:201:10: error: 'hcd_config_t' has no member named 'fifo_config'
  201 |         .fifo_config = NULL,
      |          ^~~~~~~~~~~
D:/Users/immor/CLionProjects/espressif-demo/cam/dual_cam_idf/managed_components/espressif__usb_stream/usb_host_helpers.c:201:24: warning: excess elements in struct initializer
  201 |         .fifo_config = NULL,
      |                        ^~~~
D:/Users/immor/CLionProjects/espressif-demo/cam/dual_cam_idf/managed_components/espressif__usb_stream/usb_host_helpers.c:201:24: note: (near initialization for 'hcd_config')
D:/Users/immor/CLionProjects/espressif-demo/cam/dual_cam_idf/managed_components/espressif__usb_stream/usb_host_helpers.c:217:19: error: 'hcd_config_t' has no member named 'peripheral_map'
  217 |     if (hcd_config.peripheral_map & BIT1) {
      |                   ^
```

### More Information.

I found that in IDF 5.4.1, `hcd_config_t` ([link](https://github.com/espressif/esp-idf/blob/4c2820d377d1375e787bcef612f0c32c1427d183/components/usb/private_include/hcd.h#L150)) has only one member: `intr_flags`.

YanKE01, 2025-07-28:
Hi, @yunyizhi .Thank you very much for your feedback. Currently, you can switch the IDF version to the latest [Release V5.4](https://github.com/espressif/esp-idf/tree/release/v5.4), and the issue will be fixed soon.

ziv2013, 2025-10-16:
I have tried on IDF 5.5, same issue occurs again.

[image]